azlux/log2ram #123: Question about journalctl

notDavid, 2020-08-29:
Hi there, i hope you don't mind me asking a question;

After installing log2ram (Rpi4, Ubuntu 20.04) it seems that journalctl is no longer showing any new log items. For example, running `systemctl restart lighttpd.service` and then `journalctl -u lighttpd.service` shows no log entries.
Also `journalctl -f` shows no more (new) items.

Is this expected/as designed?

If so, is there any way to use log2ram and still be able to tail the logfile with `journalctl -f`?

azlux, 2020-09-14:
Hello.
journactl use volatile storage into `/run/log/journal` . Log2ram don't interfere with this folder.

Did you have enought RAM available ? `free -h` ?

notDavid, 2020-09-15:
Thanks for your reply, I don't know what the problem was, but I do not have the device up and running anymore so I can't test it. I'll close the issue...

closed 2020-09-15 by notDavid