azlux/log2ram #124: Log2ram service not starting

bronzehedwick, 2020-09-12:
Hello, first thanks for writing this software!

I am experiencing an issue, however. I used the APT install method on a pi 3 running pihole. Everything seemed to work fine, and there were no errors in the installation. However, when I looked to see if the service was running with `df -h`, I did not find it running. I checked `sudo systemctl status log2ram`, and found this message:

    ● log2ram.service - Log2Ram
    Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset:
    Active: inactive (dead)

Any help figuring this out would be greatly appreciated. 

Thanks!

MegaV0lt, 2020-09-13:
Size of /var/log maybe too big. check with `du -h /var/log`

bronzehedwick, 2020-09-13:
Thanks @MegaV0lt ! 

Output from that command is 

```
92K     /var/log/lighttpd
4.0K    /var/log/pihole
4.0K    /var/log/samba
76K     /var/log/apt
4.0K    /var/log/private
13M     /var/log/
```

13 megs doesn't seem like too much (within the default log2ram 40 megabytes) but maybe there's something I'm missing?

MegaV0lt, 2020-09-13:
You have to check syslog. search for errors at strat or when it was installed...

azlux, 2020-09-14:
Hi,
The command `systemctl status log2ram` should give you logs after the output you give us.
Do you have somes ?

bronzehedwick, 2020-09-15:
Thanks @azlux , here's the output of that command:

```
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

Sep 13 06:25:02 raspberrypi systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Sep 14 06:25:02 raspberrypi systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Sep 15 06:25:03 raspberrypi systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
```

azlux, 2020-09-18:
That is not a good output.
Can you reboot and run the command again ?

bronzehedwick, 2020-09-18:
Absolutely @azlux . I rebooted the machine, and here's the new output:

```
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Fri 2020-09-18 23:22:50 BST; 3min 48s ago
  Process: 295 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 295 (code=exited, status=0/SUCCESS)

Sep 18 23:22:50 raspberrypi log2ram[295]: lighttpd/error.log.6.gz
Sep 18 23:22:50 raspberrypi log2ram[295]: lighttpd/error.log.7.gz
Sep 18 23:22:50 raspberrypi log2ram[295]: lighttpd/error.log.8.gz
Sep 18 23:22:50 raspberrypi log2ram[295]: lighttpd/error.log.9.gz
Sep 18 23:22:50 raspberrypi log2ram[295]: pihole/
Sep 18 23:22:50 raspberrypi log2ram[295]: private/
Sep 18 23:22:50 raspberrypi log2ram[295]: samba/
Sep 18 23:22:50 raspberrypi log2ram[295]: deleting log2ram.log
Sep 18 23:22:50 raspberrypi log2ram[295]: sent 12,959,665 bytes  received 1,933 bytes  8,641,065.33 bytes/sec
Sep 18 23:22:50 raspberrypi log2ram[295]: total size is 12,949,527  speedup is 1.00
```

I checked `df -h` and see the log2ram appearing:

```
log2ram          40M   13M   28M  32% /var/log
```

So I think rebooting the machine solved the issue.

azlux, 2020-09-18:
Indeed

closed 2020-09-18 by bronzehedwick