azlux/log2ram #125: Issue With sysstat logs ?

geminigeek, 2020-09-13:
hi,

i installed sysstat 

when i run "sar -r " i get this error "Cannot open /var/log/sysstat/sa13: Not a directory"
i added sysstat in the before in log2ram.service file also but i am still getting same error.

sysstat is working fine . kindly provide a fix/suggestion thanks.

root@ubuntu:~# service sysstat status
● sysstat.service - Resets System Activity Data Collector
     Loaded: loaded (/lib/systemd/system/sysstat.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sun 2020-09-13 21:03:46 IST; 1min 43s ago
       Docs: man:sa1(8)
             man:sadc(8)
             man:sar(1)
    Process: 2231 ExecStart=/usr/lib/sysstat/debian-sa1 --boot (code=exited, status=0/SUCCESS)
   Main PID: 2231 (code=exited, status=0/SUCCESS)

geminigeek, 2020-09-13:
hi 
i fixed it myself it seems a file was getting made '/var/log/sysstat' instead of a directory  "/var/log/sysstat/"
by creating a directory manually and removing file /var/log/sysstat i fixed the issue

closed 2020-09-13 by geminigeek

azlux, 2020-09-14:
HI,
You mean a file was in conflict with the folder creation ?

Glad to see it's fixed.

geminigeek, 2020-09-14:
hi,

"You mean a file was in conflict with the folder creation ?" Yes

yes the issue is still there

log2ram  was working fine , when i installed sysstat, sysstat could not make required folder in /var/log.
sysstat need a folder/var/log/sysstat/ but instead a file was getting created.

while i stopped sysstat , then deleted this File /var/log/sysstat  still a file was created with same name on sytstat restart
some  issue is positively there , but i resolved it by stopping sysstat and then creating an empty folder "/var/log/sysstat/" 

that fixed it.. 

my device : Odroidc2
Os: Ubuntu 20.04.1 LTS with Linux 5.8.5-meson64