azlux/log2ram #129: PATH_DISK variable question

Randomico, 2020-09-24:
I've been trying to setup multiple locations into the PATH_DISK variable inside /etc/log2ram.conf.

PATH_DISK="/var/log;/home/test/FolderInRam" works just fine but i could not make it work with a directory name with spaces like: PATH_DISK="/var/log;/home/test/Folder In Ram"

Is it possible to do that?

I've already tried the sintax below but with no success.
PATH_DISK="/var/log;/home/test/Folder\ In\ Ram"

Appreciate the help.

MegaV0lt, 2020-09-24:
log2ram script is missing a lot of " in the code. See https://www.shellcheck.net/
The make_log_dir function is propably one thing that breaks because of missing "

Randomico, 2020-09-24:
Thanks for pointing that out.
Just tried and a .hdd file is generated but only the first mountpoint (/var/log) is created, the second still missing.

MegaV0lt, 2020-09-24:
Added "some" ". Please try:
https://www.dropbox.com/s/ie1f4e2udiacv4u/log2ram?dl=0

Randomico, 2020-09-24:
Still no go with your file and the "" fixes.
Tried both methods just like below.

PATH_DISK="/var/log;/home/test/name of folder"
PATH_DISK="/var/log;/home/test/name\ of\ folder"

Using PATH_DISK="/var/log;/home/test/name_of_folder" works and both get mounted but my directory has spaces and for this one i can't change.

MegaV0lt, 2020-09-25:
Sorry, then we need more information about errors and the used system. Check your logfile

MegaV0lt, 2020-09-25:
One more thing to try:
Escape the spaces with **\040** in your mounts like:

` PATH_DISK="/var/log;/home/test/name\040of\040folder"`

Randomico, 2020-09-25:
Im using a PI with raspbian stretch. Tried with \040 but same behavior and syslog not showing mount errors.
Every time i change the the log2ram config file i reboot the machine. At PI startup, if PATH_DISK is invalid, the boot hangs for 2 minutes and 2 seconds at log2ram initiation and then finish boot successfully with only /var/log mounted as ram.

i'll test the same procedure with another distribution and see what happens, thanks for the help.

MegaV0lt, 2020-09-28:
Made some more tests. The problem is the /proc/mounts file. Spaces are encoded her (\040). I just changed teh log2ram to use the mount command to chech for the mount. Test here was working:

Debug output:
```
darkwing@darkwing-pc /media/Daten/Dropbox $ sudo bash -x log2ram start
+ SIZE=40M
+ USE_RSYNC=false
+ MAIL=false
+ PATH_DISK='/tmp/log;/tmp/log 2'
+ '[' -z '/tmp/log;/tmp/log 2' ']'
+ LOG_NAME=log2ram.log
+ case "$1" in
+ IFS=';'
+ for i in $PATH_DISK
+ PATH_FIRST_PART=/tmp
+ PATH_LAST_PART=log
+ RAM_LOG=/tmp/log
+ HDD_LOG=/tmp/hdd.log
+ LOG2RAM_LOG=/tmp/log/log2ram.log
+ make_log_dir
+ '[' -d /tmp/hdd.log ']'
+ mount --bind /tmp/log/ /tmp/hdd.log/
+ mount --make-private /tmp/hdd.log/
+ wait_for /tmp/hdd.log
+ mount
+ grep /tmp/hdd.log
+ '[' '' = true ']'
+ mount -t tmpfs -o nosuid,noexec,noatime,nodev,mode=0755,size=40M log2ram /tmp/log/
+ wait_for /tmp/log
+ mount
+ grep /tmp/log
+ syncFromDisk
+ isSafe
+ '[' -d /tmp/hdd.log ']'
+ '[' -d /tmp/hdd.log ']'
+ TP_SIZE=40M
+ '[' '' = true ']'
++ du -sh -t 40M /tmp/hdd.log/
++ cut -f1
+ '[' -n '' ']'
++ command -v rsync
+ '[' -x /usr/bin/rsync ']'
+ rsync -aXv --inplace --no-whole-file --delete-after /tmp/hdd.log/ /tmp/log/
+ tee -a /tmp/log/log2ram.log
building file list ... done
./
log2ram.log

sent 2,131 bytes  received 43 bytes  4,348.00 bytes/sec
total size is 1,983  speedup is 0.91
+ for i in $PATH_DISK
+ PATH_FIRST_PART=/tmp
+ PATH_LAST_PART='log 2'
+ RAM_LOG='/tmp/log 2'
+ HDD_LOG='/tmp/hdd.log 2'
+ LOG2RAM_LOG='/tmp/log 2/log2ram.log'
+ make_log_dir
+ '[' -d '/tmp/hdd.log 2' ']'
+ mount --bind '/tmp/log 2/' '/tmp/hdd.log 2/'
+ mount --make-private '/tmp/hdd.log 2/'
+ wait_for '/tmp/hdd.log 2'
+ grep '/tmp/hdd.log 2'
+ mount
+ '[' '' = true ']'
+ mount -t tmpfs -o nosuid,noexec,noatime,nodev,mode=0755,size=40M log2ram '/tmp/log 2/'
+ wait_for '/tmp/log 2'
+ grep '/tmp/log 2'
+ mount
+ syncFromDisk
+ isSafe
+ '[' -d '/tmp/hdd.log 2' ']'
+ '[' -d '/tmp/hdd.log 2' ']'
+ TP_SIZE=40M
+ '[' '' = true ']'
++ du -sh -t 40M '/tmp/hdd.log 2/'
++ cut -f1
+ '[' -n '' ']'
++ command -v rsync
+ '[' -x /usr/bin/rsync ']'
+ rsync -aXv --inplace --no-whole-file --delete-after '/tmp/hdd.log 2/' '/tmp/log 2/'
+ tee -a '/tmp/log 2/log2ram.log'
building file list ... done
./
log2ram.log

sent 2,482 bytes  received 43 bytes  5,050.00 bytes/sec
total size is 2,338  speedup is 0.93
```
Updated fiel is here: https://www.dropbox.com/s/ie1f4e2udiacv4u/log2ram?dl=1

@azlux You may just use the file or parts of it if you like

Randomico, 2020-09-30:
Used your file in #130 thread with PATH_DISK below and it worked, just needed a minor fix. Had to add double quotes on line 19, otherwise same error as before. Perhaps you could edit this to your commit?

`    mount -o remount,"${OPTION}" "$HDD_LOG"
`
PATH_DISK="/var/log;/home/test/name of folder"

Appreciate your help in contributing with this useful package.

MegaV0lt, 2020-09-30:
Added quotes  for function `remountOriginal`

azlux, 2020-10-01:
The PR have been merge and a new version with the fix have been released. Closing

closed 2020-10-01 by azlux