azlux/log2ram #134: Question on storage

Torontoboy51, 2020-10-23:
This is more of a newbie question than an issue ...

I'm using the default settings on Log2Ram for daily log writing at 23:55:00 time.  However, I noticed the RAM consumption before and after that time stamp is the same - no change.

I was under the impression once the data from RAM is transferred to the SD card; it should free up the space in RAM.  Is it not the case?  Please share your thoughts.

ps:  I use df -h command to verify the usage info.

zenofile, 2020-10-23:
Log2ram just shadows the non-volatile storage. The periodic transfers from volatile to non-volatile are for backup purposes to survive reboots and power outages, not for freeing up actual RAM.

You would probably need [overlay(fs)](https://www.kernel.org/doc/html/latest/filesystems/overlayfs.html) on top of `tmpfs` for that. That way it would keep only the actual modified portion of the data in RAM, until the changes are merged.

Torontoboy51, 2020-10-23:
Thanks for the explanation.
I just wanted that clarification in case I didn't properly configure the utility.  Thats all.

closed 2020-10-29 by azlux