azlux/log2ram #137: log2ram not shown in df -h command

bertiebeez, 2020-11-01:
I have the pi 4 and have installed the IOTStack in docker. As part of that install, I instaled log2ram. When I run the df -h command, it does not show up. I have uninstalled, removed the entire directory log2ram, then re-installed by the commands on this page and it runs without any errors but still not showing up  when df -h is run.
Any ideas?

azlux, 2020-11-01:
any logs ?
> The log for log2ram will be written at: /var/log/log2ram.log and available with sudo journalctl -t log2ram

staylorx, 2020-11-01:
I don't know if this will help but I ran `service log2ram status` and found I have the error: RAM disk for "/var/hdd.log/" too small. So my service wasn't starting, therefore the df -h command doesn't show it.

azlux, 2020-11-01:
That's a duplicate of https://github.com/azlux/log2ram/issues/133

bertiebeez, 2020-11-03:
Nothing in the logfile.Sent from my Samsung Galaxy Tab®|PRO

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/azlux/log2ram/issues/137#issuecomment-720111339",
"url": "https://github.com/azlux/log2ram/issues/137#issuecomment-720111339",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]