azlux/log2ram #140: Write activity is still the same, unsure how to troubleshoot

demux4555, 2020-11-17:
Hi there

I've been using log2ram for a long time on multiple RPi 3/4 devices and it works like a charm. I'm using snmp to monitor the devices, and stats on writes and written bytes to the card confirm that it's been working as it should.

I recently did a major updating of Raspbian on one device (over 200 days since last time), and now log2ram doesn't work. I've tried uninstalling it as well, and doing a clean install. 

But, I'm still seeing data being written to the card continuously (in this case it's ~250Kbyte per min). I've spent a few hours researching, and trying different things, but I'm not really getting anywhere. 

Let me know if you need more info. Any help is appreciated.

```
service log2ram status 
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Tue 2020-11-17 15:37:06 CET; 3min 41s ago
  Process: 31246 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 31246 (code=exited, status=0/SUCCESS)

Nov 17 15:37:06 K log2ram[31246]: ntpstats/sysstats.20201114.gz
Nov 17 15:37:06 K log2ram[31246]: ntpstats/sysstats.20201115.gz
Nov 17 15:37:06 K log2ram[31246]: ntpstats/sysstats.20201116.gz
Nov 17 15:37:06 K log2ram[31246]: ntpstats/sysstats.20201117
Nov 17 15:37:06 K log2ram[31246]: ntpstats/sysstats.581C0
Nov 17 15:37:06 K log2ram[31246]: private/
Nov 17 15:37:06 K log2ram[31246]: sysstat/
Nov 17 15:37:06 K log2ram[31246]: sent 103,279,557 bytes  received 3,199 bytes  68,855,170.67 bytes/sec
Nov 17 15:37:06 K log2ram[31246]: total size is 103,239,948  speedup is 1.00
Nov 17 15:37:06 K systemd[1]: Started Log2Ram.
```

```
df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  6.9G   21G  25% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           1.9G  177M  1.7G  10% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   54M  199M  22% /boot
tmpfs           383M     0  383M   0% /run/user/1000
log2ram         200M   99M  102M  50% /var/log
```

```
mount 
..
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,noatime,size=204800k,mode=755)
..
```

azlux, 2020-11-17:
Hi,
Are you able to see what file are in used ? I think about the tool `lsof +D /var/hdd.log/`

demux4555, 2020-11-17:
What am I looking for in this case? I'm not very familiar with this command.

```
lsof +D /var/hdd.log/
lsof: WARNING: can't opendir(/var/hdd.log/private): Permission denied
lsof: WARNING: can't opendir(/var/hdd.log/lightdm): Permission denied

```

```
sudo lsof +D /var/hdd.log/
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
```

demux4555, 2020-11-21:
I am unsure what has happened, but after rebooting the device several times after having it relocated in my rack, the issue seems to have resolved itself (by magic?). Literally no changes done to the device's config/files. Let me know if you need more info for further investigation. If not, you can just close this issue at your discretion.

azlux, 2021-01-07:
Srry I was busy thoses times. I'm srry I have take care of your issue earlier.
Do you still see again your issue ?

closed 2022-03-04 by azlux