azlux/log2ram #141: Installation - APT - Unable to locate package

weberc2, 2020-11-22:
I followed the installation instructions, but I'm getting an "unable to locate package" error. Is this because I'm on ARM?

```bash
ubuntu@rpi3:~$ wget -qO - https://azlux.fr/repo.gpg.key | sudo apt-key add -
OK
ubuntu@rpi3:~$ sudo apt update && sudo apt install -y log2ram
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Get:3 https://pkgs.tailscale.com/stable/ubuntu focal InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Fetched 3,997 B in 3s (1,212 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
74 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package log2ram
```

weberc2, 2020-11-22:
Never mind; just realized the `echo "deb http://packages.azlux.fr/debian/ buster main" | sudo tee /etc/apt/sources.list.d/azlux.list` was commented out...

closed 2020-11-22 by weberc2

songandsilence3, 2021-03-08:
Please re-open:

`pi@pi4-8g0:~ $ echo "deb http://packages.azlux.fr/debian/ buster main" | sudo tee /etc/apt/sources.list.d/azlux.list
deb http://packages.azlux.fr/debian/ buster main
pi@pi4-8g0:~ $ sudo apt upgrade && sudo apt install log2ram rsync  Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gconf-service gconf2-common libexiv2-14 libgconf-2-4
  libgfortran3 libgmime-2.6-0 libncurses5 libssl1.0.2 uuid-dev
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  raspberrypi-ui-mods
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package log2ram
Last login: Mon Mar  8 15:25:12 2021 from 192.168.1.147
pi@pi4-8g0:~ $ cat /etc/apt/sources.list.d/azlux.list              deb http://packages.azlux.fr/debian/ buster main
pi@pi4-8g0:~ $`

azlux, 2021-03-09:
@songandsilence3 you've forget to `apt update` to update the package source list.