azlux/log2ram #142: azlux.fr is down

mklnz, 2021-01-04:
Hi, I understand there is a manual way to install things, but to keep things consistent and easily scriptable I also prefer the apt package method. However repo server reliability comes to play here, would it be possible to host the .deb here instead?

azlux, 2021-01-04:
Having the .deb package here will not change the issue. Even with manual install, you need the apt repository's keys for auto update.

The APT source is down (I'm facing my first issue since 3 years, will be back soon ~1h)

To be honest, If I want to host the .deb into another server , it's the debian one. I'm not aware of other like bintray, so I don't trust them for now.

Azlux

mklnz, 2021-01-04:
Yes, Debian itself would be ideal of course. Thanks for the quick reply and hard work

azlux, 2021-01-04:
Ok, It's a high electrical issue. So I've switch to my backup server. Should be good now.

closed 2021-01-05 by azlux