azlux/log2ram #143: different size for different paths? (feature request?)

zadigre, 2021-01-14:
beside /var/log, I also have one other folder that is managed by log2ram. 
I need a little bit more space for this folder... so I set size to 512M.... but that means /var/log is also taking 512M. 
is there a way to set /var/log to a smaller size while keeping my other folder to 512M?

MegaV0lt, 2021-01-15:
Increasing size should be not problematic, sinc memory is only used as needed. See:
https://wiki.archlinux.org/index.php/Tmpfs#Examples

zadigre, 2021-01-19:
> Increasing size should be not problematic, since memory is only used as needed. See:
> https://wiki.archlinux.org/index.php/Tmpfs#Examples

yes I understand that... it's just my OCD that is ticked with I look at the free memory with the free command. 😉 
but also, I've seen some swap usage... which I didn't see before even if my /var/log only had 11MB and my other path was like 60MB. 
I configured log2ram with 256M instead of 512... and no swap usage so far.

azlux, 2021-01-21:
If you said you're swapping too fast, You can also check your swapiness See : https://www.howtoforge.com/tutorial/linux-swappiness/

zadigre, 2021-01-21:
oh great. thanks for the tip!
I didn't know this parameter exists. 
and a quick search on google tells me that before it was at "1" on raspbian... and it was changed to the linux default "60" in 2017. 

I changed it back to "1"... lets see how it goes in the next couple of days.

thomas725, 2022-10-21:
I too would like to see this feature, even though defining too large doesn't take up more RAM than it needs, it would be nice if we could safeguard against folders that should be tiny to grow unexpectedly and clog up our RAM.