azlux/log2ram #146: Unable to add /tmp to PATH_DISK

ronfischler, 2021-01-22:
/tmp seemed like a good candidate to add to the log2ram service, so I thought I would try this in /etc/log2ram.conf:
PATH_DISK="/var/log;/tmp"

log2ram is then unable to create /hdd.log and fails to start.

azlux, 2021-01-22:
Did you check if your system have already a tmpfs set for `/tmp` ? 
You don't need log2ram for this folder. `/tmp` is used for temporary files only.

ronfischler, 2021-01-22:
No, I don't have a tmpfs set for /tmp. You are probably right though; I don't need log2ram for /tmp, I just wanted to try minimizing write ops to my SD card I am using with my RPI. I am simply surprised it doesn't work for this folder.

ronfischler, 2021-01-22:
Looking at the log2ram script, I think the issue is on line 87: PATH_LAST_PART="${i##/*/}"

When iterating over $PATH_DISK, this line does not have the desired result when $i is a folder in the root dir. I will fiddle with the assignment made to $PATH_LAST_PART and see if I can make it work for folders such as /tmp.

azlux, 2021-04-23:
I don't remember why I didn't used basename. I need to check #SlowAnswer

dolorosus, 2022-05-05:
If you really want to add a folder in root dir you can use //foldername. 
E.G. PATH_DISK="/var/log;//tmp"