azlux/log2ram #147: start/restart log2ram.service hangs if path within PATH_DISK doesn't exist

mtklr, 2021-01-22:
Hi,

Paths in `PATH_DISK=` that don't exist seem to cause log2ram to hang when starting/restarting the service; after fixing `PATH_DISK`, it worked normally. Maybe some checking within `start)` would avoid this?

```
start)
        IFS=';'
        for i in $PATH_DISK; do

            # make sure path exists before adding it (have not tested)
            [ -d "$PATH_DISK" ] || continue

            PATH_FIRST_PART="${i%/*}"
            PATH_LAST_PART="${i##/*/}"
            RAM_LOG="$i"
            HDD_LOG="${PATH_FIRST_PART}/hdd.${PATH_LAST_PART}"
            LOG2RAM_LOG="${RAM_LOG}/${LOG_NAME}"

            make_log_dir
```

thanks for your time and for this handy utility,
matt

azlux, 2021-04-23:
Hi,
Srry I didn't check this issue earlier.
$PATH_DISK always exist, even if it's not part of the configuration. Your issue is weird.
https://github.com/azlux/log2ram/blob/master/log2ram#L5-L7

Size is for each.

closed 2022-03-04 by mtklr

azlux, 2022-03-04:
F--- I forgot this issue.
It's was a one line work.
done... one year later

mtklr, 2022-03-09:
That's fine - Thanks so much for getting to it, I appreciate it! :)