azlux/log2ram #148: How to check if it works correctly?

tomini, 2021-02-06:
Hello,
I have few concerns. In **Web UI/flutterhole** (Android app connecting via API) the **queries are still same**. Updating within milliseconds.
This s**cript is rerouting/logging queries to RAM** and not just logs of the system info, failures and stuff like that. **Do I understand it correctly?**
**Log2RAM shows up in** mount and **_df -h_**, but I **changed it to 80MB** instead of 40MB and **it still shows 40MB**. **What do I need to reboot**/restart **to apply** these changes? I didn't want to command it to reboot via PuTTY, because **I'm scared that all logs will vanish**.
**Does reboot via PuTTY or Pihole Web UI** make this script **save those queries to microSD?** **How to check that they** indeed **do save** and not vanish in RAM...
**Can this easily script fail in some way?** **In which way** and how to **prevent it?** I can only think of an update for Pihole that somehow breaks this script...and I will be stuck without query feedback.
How **can I view the write history to my microSD?** So I could confirm that everything is working fine.

**If there are any trade-offs and things to know - I will be glad and really grateful for knowing them.**

Thank you so much! <3

azlux, 2021-04-23:
- What do I need to reboot/restart to apply these changes?
-> Yes
- Does reboot via PuTTY or Pihole Web UI make this script save those queries to microSD?
-> The event 'reboot/shutdown' force log2ram to flush on SD the log
- How to check that they indeed do save and not vanish in RAM...
-> use `service log2ram reloead` , this flush the log on SD manually, you can check `/var/hdd.log` if all you logs are here.
- How can I view the write history to my microSD?
-> `sudo journalctl -t log2ram` are log2ram logs.
- Can this easily script fail in some way?
-> main case is if your log is bigger than max configuration size. Configure logrotate to avoid big log.

bentkowski, 2022-05-09:
Hi. 
Im having same problem
im not sure if its working

pi@rasp3:/var/log $ systemctl status log2ram
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
     Active: active (exited) since Mon 2022-05-09 18:24:03 CEST; 20min ago
    Process: 230 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
   Main PID: 230 (code=exited, status=0/SUCCESS)
        CPU: 994ms

Warning: journal has been rotated since unit was started, output may be incomplete.

This active(exited) status is confusing