azlux/log2ram #150: Unable to execute bash script

jerryryb, 2021-02-24:
Hi,
In PATH_DISK I added another path, then try to run some script e.g. ./test.sh and I got error
`-bash: ./test.sh: Permission denied`

```
ubuntu@ubuntu:~/ramplace$ ls -l test.sh
-rwxrwxr-x 1 ubuntu ubuntu 20 Feb 24 18:48 test.sh

```
```
ubuntu@ubuntu:~/ramplace$ cat test.sh
echo "sample text"

```
My system:
raspberry pi
Linux ubuntu 5.4.0-1028-raspi #31-Ubuntu SMP PREEMPT Wed Jan 20 11:30:45 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

Jerry

jerryryb, 2021-03-04:
When I deleted **noexec** below then it works as I expected
`if [ "$ZL2R" = true ]; then
    createZramLogDrive
    mount -t ext4 -o nosuid,noexec,noatime,nodev,user=log2ram "/dev/zram${RAM_DEV}" "$RAM_LOG"/
else
    mount -t tmpfs -o "nosuid,noexec,noatime,nodev,mode=0755,size=${SIZE}" log2ram "$RAM_LOG"/
fi`

azlux, 2021-03-05:
Hi,
Sorry, I didn't saw your issue earlier.
Indeed, the mount is noexec.
But I don't see the point to have executable files into the log2ram directory. The purpose of an exec isn't to be modified every minutes.

closed 2021-10-12 by azlux