azlux/log2ram #151: Does this work with Ubuntu?

influx3k, 2021-03-02:
Any way to get this to work with Ubuntu 20.04? I’ve tried twice and could not get it to work. Did not see log2ram with df-h

azlux, 2021-03-02:
Hi,
- You have ubuntu on a sd card ?
- This should work. Any error into `/var/log/log2ram.log`  or `sudo journalctl -t log2ram` ?

influx3k, 2021-03-03:
Yes, Ubuntu 20.04 and 20.10 are supported on the Pi now.

Got cat: `/var/log/log2ram.log: No such file or directory` and 
```
-- Logs begin at Thu 2021-02-04 00:20:16 EST, end at Tue 2021-03-02 19:24:44 EST. --
-- No entries --
```

influx3k, 2021-03-03:
I figured it out, and a note to anyone else who may find this.... Log2Ram will not start (and hence show anything in `df -h)` if the logs in /var/log are already larger than the SIZE variable set in /etc/log2ram.conf.  You can either set SIZE larger, trim down the size of the logs, or a combination of both.  For me, the size of the journal was the main culprit; racking up over 3GB.  You can use `journalctl --vacuum-size=100M` to trim down the size of the journal to something more manageable.  Set SIZE to 128M or 256M, save and reboot and you should see Log2Ram appear in `df -h`.

This should be added in the readme for future users.

azlux, 2021-04-23:
thank for the solution, I wasn't expecting an journalctl limitation here. Log2ram report when the size isn't good.

caheredia, 2021-06-30:
> This should be added in the readme for future users.
I was about to say the same thing

azlux, 2021-09-06:
See https://github.com/azlux/log2ram#existing-content-in-varlog-too-large-for-ram

closed 2021-09-06 by azlux