azlux/log2ram #152: Docker logs

rafis, 2021-03-06:
How to store Docker logs in log2ram disk?

TomyCesaille, 2021-04-13:
Following <https://pimylifeup.com/raspberry-pi-log2ram/>, you can add additional paths into log2ram.

azlux, 2021-04-23:
It's a hard question.
Logs are part of images on docker and/or part of log drivers on docker system.
On my side, I disable them to avoid logs on docker side, and my nginx logs are a mount point from the host where a can manage the location.
Maybe the good solution is to use the log-driver [journald or syslog](https://docs.docker.com/config/containers/logging/configure/#supported-logging-drivers) to manage size, location and log2ram path.

ViRb3, 2021-05-11:
My solution is to dump all logs into journald from Docker:
```yml
  logging:
    driver: journald
    options:
      tag: "container-{{.Name}}"
```

marcelstoer, 2021-08-15:
This is a tough one and I am also looking for a better solution.

I currently have `/var/lib/docker/containers` in log2ram as well. However, I made the foolish mistake of not checking beforehand how large the `/var/lib/docker/containers/<container>/<container>-json.log` files grow. Sure enough the combined log files (4 containers) wouldn't fit into `SIZE` configured for log2ram. I increased it to 512M but that's of course just a temporary fix.

Ideally I think a rotate-like feature in log2ram could address this. In such a mode the RAM content would be appended to the actual log file upon syncing _and then discarded_. Hence, log2ram would only retain the messages added since the last sync.

xtvdata, 2021-08-15:
Why not to just add an entry in /etc/rsyslog.d/ and rotate the logs as needed?

If the logs are indeed huge (and really needed with that detail) the only real solution is to get a lot more RAM and use log2ram, or just go for an ssd… or in general if you need to manage a very large amount of logs, there are other options much better than plain text files such as logging to a separate log server with something such as loki (possibly with an SSD drive).

Personally I prefer to use log2ram with a relatively short log rotation history on servers (because I don’t like to waste much RAM on logs) and store long term logs separately on Loki (at home) or on an Elasticsearch cluster (at office).

marcelstoer, 2021-08-16:
For now I configured log rotation only for the Docker logs using [their built-in mechanism for the JSON files](https://docs.docker.com/config/containers/logging/json-file/).