azlux/log2ram #153: Install Error on 32-bit Raspberry Pi OS (Raspbian)

fracture-point, 2021-03-25:
I followed the apt instructions for installation, however I get the errors below:
```
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
...
installed sysdig-dkms package post-installation script subprocess returned error exit status 10
```
Here's the full log of installation activity:

```
pi@raspberrypi:~ $ echo "deb http://packages.azlux.fr/debian/ buster main" | sudo tee /etc/apt/sources.list.d/azlux.list
deb http://packages.azlux.fr/debian/ buster main
pi@raspberrypi:~ $ wget -qO - https://azlux.fr/repo.gpg.key | sudo apt-key add -
OK
pi@raspberrypi:~ $ sudo apt update
Get:1 http://httpredir.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                                                                    
Hit:3 https://download.docker.com/linux/raspbian buster InRelease                                                                                                     
Get:4 http://archive.raspberrypi.org/debian buster InRelease [32.9 kB]                                                                                                
Get:5 http://packages.azlux.fr/debian buster InRelease [3,983 B]                                                                                                      
Ign:6 https://ookla.bintray.com/debian buster InRelease                              
Get:7 https://ookla.bintray.com/debian buster Release [4,387 B]
Get:8 http://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Get:9 http://packages.azlux.fr/debian buster/main armhf Packages [2,893 B]
Fetched 463 kB in 3s (146 kB/s)                                    
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
pi@raspberrypi:~ $ sudo apt install log2ram
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  log2ram
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 4,288 B of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://packages.azlux.fr/debian buster/main armhf log2ram all 1.6.0 [4,288 B]
Fetched 4,288 B in 0s (17.5 kB/s)
Selecting previously unselected package log2ram.
(Reading database ... 99598 files and directories currently installed.)
Preparing to unpack .../archives/log2ram_1.6.0_all.deb ...
Unpacking log2ram (1.6.0) ...
Setting up sysdig-dkms (0.21.0-1) ...
Removing old sysdig-0.21.0 DKMS files...

------------------------------
Deleting module version: 0.21.0
completely from the DKMS tree.
------------------------------
Done.
Loading new sysdig-0.21.0 DKMS files...
It is likely that 5.10.17-v7l+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
Consult /var/lib/dkms/sysdig/0.21.0/build/make.log for more information.
dpkg: error processing package sysdig-dkms (--configure):
 installed sysdig-dkms package post-installation script subprocess returned error exit status 10
Setting up log2ram (1.6.0) ...
Created symlink /etc/systemd/system/sysinit.target.wants/log2ram.service → /etc/systemd/system/log2ram.service.
Created symlink /etc/systemd/system/timers.target.wants/log2ram-daily.timer → /etc/systemd/system/log2ram-daily.timer.
#####         Reboot to activate log2ram         #####
##### edit /etc/log2ram.conf to configure options ####
Errors were encountered while processing:
 sysdig-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Note that there is no chroot as implied in the script - I'm trying to install this directly on the host OS.

fracture-point, 2021-03-25:
I also just got the same error while trying to REMOVE the failed installation:

```
pi@raspberrypi:~ $ sudo apt-get remove log2ram
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  log2ram
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 99607 files and directories currently installed.)
Removing log2ram (1.6.0) ...
Removed /etc/systemd/system/sysinit.target.wants/log2ram.service.
Removed /etc/systemd/system/timers.target.wants/log2ram-daily.timer.
Setting up sysdig-dkms (0.21.0-1) ...
Removing old sysdig-0.21.0 DKMS files...

------------------------------
Deleting module version: 0.21.0
completely from the DKMS tree.
------------------------------
Done.
Loading new sysdig-0.21.0 DKMS files...
It is likely that 5.10.17-v7l+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
Consult /var/lib/dkms/sysdig/0.21.0/build/make.log for more information.
dpkg: error processing package sysdig-dkms (--configure):
 installed sysdig-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 sysdig-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I'm afraid to reboot at this point...

xtvdata, 2021-03-25:
Hmm, from your posts it seems that `log2ram` is installed properly and also its removal doesn't raise any issue.

On the other hand it looks like you have issues with a somewhat broken installation of the package `sysdig-dkms` (which is a dynamic kernel module) not related to `log2ram`.

> **Package:** sysdig-dkms
 **Description:** system-level exploration and troubleshooting tool - kernel source
 Sysdig instruments your physical and virtual machines at the OS level
 by installing into the Linux kernel and capturing system calls and
 other OS events. Then, using sysdig's command line interface, you can
 filter and decode these events in order to extract useful information
 and statistics.
 .
 Sysdig can be used to inspect live systems in real-time, or to generate
 trace files that can be analyzed at a later stage.
 .
 This package contains the source for the kernel module of sysdig.

If you check the log you posted you can see that when you run `apt` to install `log2ram` you get the lines:
```
The following NEW packages will be installed:
  log2ram
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
```

The "1 newly installed" is `log2ram`, while the `1 not fully installed or removed` is `sysdig-dkms` which was installed (partially) before you installed `log2ram` (otherwise `sysdig-dkms` would be listed in the NEW packages too).

**Suggestion:** before installing _any_ other package you should investigate and resolve the issues in your installation of `sysdig-dkms` (or any other package that installed that as dependency).

fracture-point, 2021-03-25:
Thanks @xtvdata , you're right that these are apparently unrelated (I thought that package may have been getting modified as part of the install).  Per https://raspberrypi.stackexchange.com/a/116510, it looks like sysdig is a package coming through from the Debian port that doesn't work on Raspbian.  I removed it with apt, reinstalled log2ram, and on reboot everything looks great.  Thanks!

closed 2021-03-25 by fracture-point