azlux/log2ram #155: Install scripts in `/usr/local/sbin` instead of `/usr/local/bin`

gprst, 2021-04-23:
According to [this StackOverflow answer](https://askubuntu.com/questions/308045/differences-between-bin-sbin-usr-bin-usr-sbin-usr-local-bin-usr-local#308048) referring to the [Filesystem Hierarchy Standard (FHS) for Linux](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard), binaries (and I guess, scripts) that _require root priviledges_ have to be in `sbin` directories (such as `/sbin`, `/usr/sbin` and, in our case, `/usr/local/sbin`), while `bin` directories are for general system-wide binaries (think `cat`, `ls`), that don't require root priviledges.

Since `log2ram` require root priviledges, I think its scripts should be installed in `/usr/local/sbin` instead of `/usr/local/bin`. Since this is quite a breaking change, I didn't make a PR, so that you can decide for yourself ! (Thank you very much for your software, by the way, it is awesome 😄)

azlux, 2021-04-23:
This is a big question that has been discussed many times with friends.
Log2ram is managed by your packet manager (for debian/ubuntu) so the path should be `/usr/sbin`
But, it's install manually on many system, so for manual install script, the path should be `/usr/local/sbin`.

But on `man hier`:
```
/usr/local/sbin
    Locally installed programs for system administration.
/usr/local/bin
    Binaries for programs local to the site.
```
I didn't saw `log2ram` as a system administration program first, when I start writing the program, it is now. 
Hard to tell, choice have been made, and it hard to change without effect.
I don't think I will change that

gprst, 2021-04-23:
This is an answer I expected 😉 You can close the issue if you want 👍

closed 2021-04-23 by azlux