azlux/log2ram #158: Document how to set up systemd-journal to keep a constant size

raek, 2021-06-09:
Hi! Thanks for this great package! It really solves a common problem quite elegantly.

When I installed log2ram it turned out I had about 350 megabytes of log files in /var/log/journal (the pi had been running for a few months...). The log2ram.service failed since the default partition size is 40 megabytes. I needed to configure journald to only use 20 megabytes. I did it like this:

* sudoedit /etc/systemd/journald.conf # edit line: "SystemMaxUse=20M"
* sudo restart systemd-journald # this shrunk my existing logs to slightly over 20M.
* sudo reboot # after this the log2ram.service succeeded

How about adding some instructions for how to do this in the readme? What do you think?

raek, 2021-06-09:
I just realized that the default journald behavior is to use a fraction of the available space in /var/log for the logs (10%?). Maybe the default settings would have worked just fine if I had shrunk my journal down to fit in the default 40 megabyte tmpfs partition...

azlux, 2021-06-14:
Do you want to add some instructions on the Readme with a PR ? Or I can do it myself later this week.

raek, 2021-06-15:
I'm working on the instructions tonight. I'll submit a pull request when I'm done!

raek, 2021-07-28:
Closing this since the documentation change is done.

closed 2021-07-28 by raek