azlux/log2ram #163: RAM disk for "/var/hdd.log/" too small. Can't sync. (but system is reporting:  4.0K /var/hdd.log) - Pop!_OS 21.04

MostHated, 2021-07-10:
Hey there,
    I came across this project today when looking for, well, something that did exactly what this does so I decided to give it a try as this new firewall software I was having a go at seems to write nonstop logs to disk on every network action for every application, it seems. [(Portmaster)](https://github.com/safing/portmaster) I figured that instead of subjecting my SSD's to all those writes constantly, why not log it to ram first, then just dump that in one go. 

  That said, I installed it, cleaned up my logs a big before hand to get the size down to a reasonable amount and restarted and went to check if it was running, which was when I saw the message of:

```sh
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2021-07-10 16:08:56 CDT; 19min ago
    Process: 1377 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
   Main PID: 1377 (code=exited, status=1/FAILURE)

Jul 10 16:08:56 pop-os log2ram[1377]: ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync.
Jul 10 16:08:56 pop-os log2ram[1424]: /usr/local/bin/log2ram: 45: mail: not found
```

I was under the impression that it only dealt with /var/log/*, and I can't even say I remember seeing /var/hdd.log before either? Checking out the folder, it reported back the following:

```sh
zsh: sudo du -hs /var/h* | sort -h | tail -n 3                                                                   mosthated@pop-os
4.0K    /var/hdd.log
```

Any recommendations on what I might need to check/do? My system is running Pop!_OS 21.04.

Thanks,
-MH

MostHated, 2021-07-11:
Nevermind about it, it seems I have got it working.

```sh
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sun 2021-07-11 17:48:10 CDT; 49s ago
    Process: 1602947 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
   Main PID: 1602947 (code=exited, status=0/SUCCESS)
```

DietShasta, 2021-07-15:
I've actually just run into this same exact problem on my Raspberry PI 4, but I am unsure how to fix it.

```
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-07-15 08:42:21 PDT; 3min 3s ago
  Process: 299 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 299 (code=exited, status=1/FAILURE)

Jul 15 08:42:21 homebridge log2ram[299]: ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync.
Jul 15 08:42:21 homebridge log2ram[299]: /usr/local/bin/log2ram: 45: /usr/local/bin/log2ram: mail: not found
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

TupaNegreiros, 2021-08-18:
Same error here. Ubuntu for Raspberry Pi (64bits). Empty /var/hdd.log folder, not even inside /var/log.

How did you solve it?

MostHated, 2021-08-18:
I just made it bigger, something like 400mb and removed as much as I could and set limits.

TupaNegreiros, 2021-08-19:
My Ubuntu /var/log has about 150mb, I set log2ram to 160mb and it started.
But Raspbian /var/log has almost 900mb and only 1Gb of RAM.

@azlux Should it store on RAM even old rotated logs? (daemon.log.1, syslog.1, *.gz, *.[0-9])

xtvdata, 2021-08-19:
> My Ubuntu /var/log has about 150mb, I set log2ram to 160mb and it started.
> But Raspbian /var/log has almost 900mb and only 1Gb of RAM.
> 
> Should it store on RAM even old rotated logs? (daemon.log.1, syslog.1, *.gz, *.[0-9])

It is technically not possible to sync selectively, since the Ram disk is mounted over the root disk at the /var/log location and synced back to the root disk periodically. When the RAM disk is mounted you can only see the content of the RAM disk, thus a selective sync would break the standard Linux log rotation mechanism (and possibly even something else).

However you could eventually change logrotate’s configuration to _rotate_ some or all of the old logs to a different archive directory which would not be synced to the RAM disk (not under /var/log). But that is out of the log2ram’s scope.

Anyway 900MB of logs is quite significant for a Raspberry Pi. Are you sure that you don’t have some applications installed by you on top of the vanilla install, that are not managed by log rotation properly and are generating an ever growing log? How big are the 3 largest files you have in /var/log (or in one of the subdirectories)?
If those are not rotated during the night you are missing some system the logrotate.

Really, 900MB of textual logs is really huge for a non large business/production system. Consider that in 1MB of plain text you can, more or less, fit in a 500 pages novel or 10,000 events described with a 100 characters messages. Moreover, if rotation is configured properly, most of the files in /var/log should be already compressed (with an average efficiency of 4-5:1)… Are you analyzing all of that info?
If logs are that large for debug purposes (e.g. an alpha-test/development server) you can rotate more frequently and probably keep just a few days of history to check the tests results.

I’d recommend to check what is generating so much logging, confirm that you really need that level of detail and modulate at the origin log level based on you needs, otherwise, if you really need that much logging, you should consider a different log strategy (e.g. a dedicated log server), a structured log server will indeed serve you much better than just plain text files (in this case log2ram could help in keeping the low the server load).

bubuntux, 2022-02-18:
i ran into the same issue and fixed :
checking the size of the current log folder 
`du -h /var/log/ | tail -n1`

change the config SIZE in /etc/log2ram.conf to something grater than the value of the size of the log folder 

restart the service

azlux, 2022-02-23:
@TupaNegreiros log2ram store everything. Storing only new log was tricky for this kiss project.

Blindfreddy, 2022-02-28:
Followed instructions for installation with apt. I get the same problem on two different pis:  

```
Feb 28 08:20:47 rpi1 log2ram[314]: ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync.
Feb 28 08:20:47 rpi1 log2ram[314]: /usr/local/bin/log2ram: 45: /usr/local/bin/log2ram: mail: not found
```

strangely, the file is actually very small:

```
$ ll /var/hdd.log/
total 0
```

increased size to 500MB in conf but same problem.

azlux, 2022-03-01:
You're right, I will make a better error message.

About
> Feb 28 08:20:47 rpi1 log2ram[314]: /usr/local/bin/log2ram: 45: /usr/local/bin/log2ram: mail: not found
You have set MAIL to `true` that why log2ram try to send one.

Blindfreddy, 2022-03-07:
Ok great.  And thanks for the hint about Mail, I will change the value in my config file.

It might also be useful to include some instructions and/or links on how to reduce the log size before activating log2ram in the documentation.

thewriteway, 2022-03-26:
I couldn't get the service to start due to my Pi having excessively large log files.

To fix that problem I removed the legacy ones and truncated the existing log files this way:
sudo rm -f *.gz
sudo truncate -s100 *.log
sudo truncate -s100 *.log.*

xtvdata, 2022-03-26:
> I couldn't get the service to start due to my Pi having excessively large log files.
> 
> firstly I tried the journal reduction: sudo journalctl --vacuum-size=16M That wasn't sufficient so I removed the other legacy logs and truncated the active ones from the log directory:
> 
> sudo rm -f *.gz sudo truncate -s50 *.log

Well, just "truncating" the logs will not solve your issue.
It will improve the situation, but only for a few time, then eventually `/var/log` will grow too large again.
If you had to do that it means that you have a problem with log verbosity (e.g. debug) and/or with log rotation.

Thus, you should reduce verbosity of your apps and/or fine tune the configuration log _logrotate_ (see other discussions on the log size topic).

tehstu, 2022-05-18:
Using the official imaging tool, I just installed Raspberry Pi OS Lite 64bit on an RPi 3b and, after updating packages, installed log2ram. I was presented with the same error as noted on this issue:

`ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync`

I eventually determined it was because `/var/log` was 41M, slightly too large for the default `SIZE=40M`. I resolved it editing `/etc/log2ram.conf` and setting `SIZE=80M`, and then rebooting.

Strangely, this was not the case when I installed Ubuntu 22.04 LTS on the same device. I guess a fresh install of Raspberry Pi OS tends to log a bit more?

DarkSkull91, 2022-06-19:
> 

I face this same problem on a fresh install of Raspberry Pi OS. Increasing the size to `SIZE=80M` solves the problem.

heutger, 2023-01-02:
Same for me with a fresh lite Raspberry Pi OS 64 bit, seems like new installations take more log file size and default setting is too low for them. With 32 bit and Buster it worked, but 64 bit and Bullseye it didn't.

maximelafarie, 2024-01-19:
Even if it's strange, `SIZE=80M` solves the problem for me too.

bmn001, 2024-04-19:
It seems like the install process should check current logfile sizes and offer to purge excess before installing to avoid this error, which seems now to be nearly-default behavior for installs across several OSs, no?

theking2, 2025-03-29:
> I just made it bigger, something like 400mb and removed as much as I could and set limits.

what did you make "bigger"?

theking2, 2025-03-29:
In my case the journal of a system running dnsmasq, ssh jumpstation and sshguard over a two year period generated 1.7G of systemd-journal data. So I had to first reduce that to a manageble size with:

```sh
journalctl --vacuum-size=128M
```

After that reload the journal:

```sh
systemctl restart systemd-journald
```

Also make sure the journal size will be in a reasonable size by editing `/etc/systemd/journald.conf` and uncomment and modify the line

```sh
SystemMaxUse=128M
```

Now I was able to start the service with 

```sh
systemctl start log2ram
```

and check that it ran with 

```sh
systemctl status log2ram
```

Also check with `iostat 5` to make sure there are no writes on a physical device