azlux/log2ram #167: log2ram rollover syntax

KingArrkinian, 2021-08-15:
Hello,

Thank you for this great program! Probably going to save quite a few months on my Pi.

I have a question/problem regarding the standard syntax for Timer:
**[Timer]
OnCalendar=*-*-* 23:55:00**

This was the setup in log2ram-daily.timer, but it seemed to never work. Despite working on a very low-intensity server, the log reached 255Mb of usage(and kept giving me failures because it was limited at 200Mb).

I'd like to get the reference for this syntax, since it isn't standard Cron. Perhaps putting it in the README might be useful.

For now I've changed it to OnCalendar=weekly, with hopes that it'll rollover properly this time.

KingArrkinian, 2021-08-15:
Edit: read in the ongoing PR that the syntax is from systemd.timer.

Maybe just add that fact and a link to https://www.man7.org/linux/man-pages/man7/systemd.time.7.html in your README?

xtvdata, 2021-08-15:
> Despite working on a very low-intensity server, the log reached 255Mb of usage(and kept giving me failures because it was limited at 200Mb).

Hi,
The timer will only sync the ram to the disk, as log2ram does not clear the logs, ever.

If the log partition is filling up, you need to take a look at the log rotation which is the proper way to limit the amount of logs preserved.

Possibly you have some application logging which is not managed by rsyslog/syslog-ng, or you have some application’s log level set to debug (generating a huge amount of log entries).

closed 2021-09-30 by KingArrkinian