azlux/log2ram #175: Log2ram failed to start on boot

SodaWithoutSparkles, 2021-10-18:
Device: raspberry pi 4 2GB 
kernal: 5.10.63-v7l+

I noticed that my pi took a very long time to reboot. Then I checked that boot.log and found that log2ram failed to start. When I manually restart log2ram by 
```
sudo systemctl restart log2ram
```
it worked totally fine.

Then I removed log2ram via apt method, then installed it again. The problem is gone.

It might be due to a update I done 3 days ago, since I havent rebooted since then.

Anyone who thinks that their raspberry pi failes to boot after reboot or just failes to boot should check if they have log2ram installed. If so, wait for about 5 minutes after power on to see if it boots up. If that is the case, try doing the following:

```
sudo apt update; sudo apt remove log2ram && sudo apt install log2ram
```

SodaWithoutSparkles, 2021-10-19:
I manage to get it again, somehow

```
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Tue 2021-10-19 17:23:41 HKT; 3min 14s ago
  Process: 274 ExecStart=/usr/local/bin/log2ram start (code=killed, signal=TERM)
 Main PID: 274 (code=killed, signal=TERM)

10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/bootstrap/fonts/glyphicons-halflings-regular.woff
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/bootstrap/fonts/glyphicons-halflings-regular.woff2
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/bootstrap/js/
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/bootstrap/js/bootstrap.min.js
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/font-awesome/
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/font-awesome/js/
10月 19 17:23:42 raspberrypi log2ram[274]: admin/style/vendor/font-awesome/js/all.min.js
10月 19 17:23:42 raspberrypi log2ram[274]: colour/
10月 19 17:23:42 raspberrypi log2ram[274]: colour/Black.png
10月 19 17:23:42 raspberrypi log2ram[274]: colour/black.html
```

azlux, 2021-10-19:
It's look like the copy stop somewhere.
Can you try to launch the copy manually ? 
```
/usr/local/bin/log2ram write
```

XhmikosR, 2021-11-04:
I think I'm also hitting this but how can I verify it? I mean, I do get like 5-min bootup times after a reboot, but it doesn't happen consistently.

SodaWithoutSparkles, 2021-11-04:
@XhmikosR the most simple solution is to check the status of log2ram right after reboot by

    sudo systemctl status log2ram

If it said `active (exited)` its should be fine.

Make sure that you dont have any script that restarts log2ram. If any they should be in crontabs. If you dont remember setting any of those, the chances are you dont have any.

Or you could check the `/var/log/boot.log`, it should contain important data. Check for any lines that have at least one red charactors, or `grep -i` it with `log2ram` or `fail`.

SodaWithoutSparkles, 2021-11-05:
You may need to use sudo to read boot.log

Or you could try to reinstall log2ram, but be reminded that all logs, including the archived ones would be gone. Make sure you dont save anything important there.

XhmikosR, 2021-11-07:
I believe my issue was due to https://github.com/RPi-Distro/raspberrypi-net-mods/issues/6. So, I was getting aprox. 90-100 seconds boot delay due to that and not due to log2ram. Sorry for the mixup, I was trying to pinpoint the cause :)

SodaWithoutSparkles, 2021-11-19:
@azlux I have an idea. Maybe log2ram was configured to start before zram, so it tried to write to a non-existing drive?

azlux, 2021-11-19:
Can you check the zram service on your system ? 
On the log2ram service, systemd start log2ram before a service called `zram-swap-conf.service`
https://github.com/azlux/log2ram/blob/master/log2ram.service#L4

SodaWithoutSparkles, 2021-11-19:
```
[Unit]
Description=Log2Ram
DefaultDependencies=no
Before=basic.target rsyslog.service syslog-ng.service syslog.target systemd-journald.service sysinit.target shutdown.target zram-swap-conf.service apache2.service lighttpd.service
After=local-fs.target
Conflicts=shutdown.target reboot.target halt.target
RequiresMountsFor=/var/log /var/hdd.log
IgnoreOnIsolate=yes

[Service]
Type=oneshot
ExecStart= /usr/local/bin/log2ram start
ExecStop= /usr/local/bin/log2ram stop
ExecReload= /usr/local/bin/log2ram write
TimeoutStartSec=120
RemainAfterExit=yes

[Install]
WantedBy=sysinit.target
```

It looks like this now. How should I modify it? I dont quite understand...

azlux, 2021-11-19:
That was not the question.
Do you have an answer with the command : `systemctl --type=service | grep zram`

clavay, 2021-12-07:
I had the same problem.
Reinstalling log2ram solve the problem.
Before and after reinstalling this (systemctl --type=service | grep zram) answer nothing.
I also use log2ram for another directory.

SodaWithoutSparkles, 2021-12-07:
Strange. Both my zram and log2ram is working now, but that command I just ran give me this result:

sudo systemctl --type=service | grep zram
```
  zram-setup@zram0.service                                    loaded active exited  Setup zram based device zram0
  zram-setup@zram1.service                                    loaded active exited  Setup zram based device zram1
  zram-setup@zram2.service                                    loaded active exited  Setup zram based device zram2
  zram-swap.service                                           loaded active exited  zram swap service
```
```
pi@raspberrypi:~ $ zramctl
NAME       ALGORITHM DISKSIZE  DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram2 lz4           2.1G 32.7M    9M 13.4M       4 [SWAP]
/dev/zram1 lz4           200M 33.8M   13M 21.8M       4 /var/www/html
/dev/zram0 lz4           200M 98.3M 26.8M   53M       4 /var/log
```

azlux, 2021-12-09:
This fix should help with your issue : https://github.com/azlux/log2ram/commit/6ee7907b80cea061e0ff0b0f9479bd304f7e977c

To test the fix before I release next version, you can make the same modification than the commit on this file : /etc/systemd/system/log2ram.service

SodaWithoutSparkles, 2021-12-12:
@azlux no, that does nothing.

SodaWithoutSparkles, 2021-12-13:
zram must be causing some problem.

clavay, 2021-12-14:
I think it's not related to zram but having more than one directory in log2ram conf.

SodaWithoutSparkles, 2021-12-14:
Maybe both?

```
zram     Num of dir   problems
No         1                    No
No         2                    Not tested
Yes        1                    Not tested
Yes        2                    Yes

```

Triggering the bug is not entirely obvious yet. I suppose it was due to some apt updates? Sometimes after a apt update I got this bug, but I seldom reboots anyway, so I am not sure. I think it was waiting for something for too long, so including the time for copying files exceeds the 120s limit? If so what is hanging it off? Mounting the disks?

Maybe it log2ram isnt designed for more than 1 dir, so the service file does not add the second dir?

SodaWithoutSparkles, 2021-12-14:
Looking at the service file, shouldnt log2ram starts after zram? If so, why that isnt marked as `wanted`? Or, why `zram-swap-conf` is marked in the after section, so it is started after log2ram? I dont quite understand.

un99known99, 2022-01-01:
Happy new 2022 first and many thanks for the great tool - I am really addicted to it :)

Wanted to ask if any updates here?
I am really frustrated as I am heavyly rely on log2ram, dont know since when log2ram is not starting due to timeout, staring time more than 2min instead of seconds, if I start log2ram manually it runs, but if I create directories in /var/log, they are gone afterwards (seems they are not written to disk), here my logs:

Jan  1 12:20:38 raspberrypi systemd[1]: Startup finished in 5.246s (kernel) + 2min 39.356s (userspace) = 2min 44.603s.

root@raspberrypi[~] # systemctl status log2ram
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Fri 2021-12-31 11:42:49 CET; 23h ago
  Process: 276 ExecStart=/usr/local/bin/log2ram start (code=killed, signal=TERM)
 Main PID: 276 (code=killed, signal=TERM)

 log2ram         144M   33M  112M  23% /var/log
=> NO /var/hdd.log

Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades-shutdown.log.2.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades-shutdown.log.3.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades-shutdown.log.4.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades-shutdown.log.5.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades-shutdown.log.6.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades.log
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades.log.1.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades.log.2.gz
Dec 31 11:42:50 raspberrypi log2ram[276]: unattended-upgrades/unattended-upgrades.log.3.gz
Jan 01 11:21:39 raspberrypi systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.

dmesg
[  130.870620] systemd[1]: log2ram.service: Start operation timed out. Terminating.
[  130.872632] systemd[1]: log2ram.service: Main process exited, code=killed, status=15/TERM
[  130.873040] systemd[1]: log2ram.service: Failed with result 'timeout'.
[  130.875143] systemd[1]: Failed to start Log2Ram.

start manually
root@raspberrypi[~] # systemctl status log2ram
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2022-01-01 11:25:00 CET; 4s ago
  Process: 2489 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 2489 (code=exited, status=0/SUCCESS)

log2ram         144M   33M  112M  23% /var/hdd.log
log2ram         144M   33M  112M  23% /var/log

apt list log2ram
log2ram/oldstable,now 1.6.0 all [installed]

Problem is I cant fresh install log2ram, afterwards most of the logs in /var/log are gone

My log2ram.conf:
SIZE=144M
USE_RSYNC=true
MAIL=true
PATH_DISK="/var/log"
ZL2R=false
33 COMP_ALG=lz4
39 LOG_DISK_SIZE=360M

Help here is really needed and appreciated

Thx, 
Mike

un99known99, 2022-01-01:
Jan 1 12:20:38 raspberrypi systemd[1]: Startup finished in 5.246s (kernel) + 2min 39.356s (userspace) = 2min 44.603s

disabling server log2ram:
Jan  1 13:02:44 raspberrypi systemd[1]: Startup finished in 5.269s (kernel) + 38.442s (userspace) = 43.712s.

un99known99, 2022-01-04:
add. info:
I can manually start log2ram via sudo systemctl start log2ram, aferwards it is running:

log2ram         144M   35M  110M  24% /var/hdd.log
log2ram         144M   35M  110M  24% /var/log

so what is causing this startup crash??????

[   10.819898] systemd[1]: Starting Log2Ram...
[  130.975678] systemd[1]: log2ram.service: Start operation timed out. Terminating.
[  130.976959] systemd[1]: log2ram.service: Main process exited, code=killed, status=15/TERM
[  130.977368] systemd[1]: log2ram.service: Failed with result 'timeout'.
[  130.979386] systemd[1]: Failed to start Log2Ram.

SodaWithoutSparkles, 2022-01-04:
Timeout apprently. Maybe log2ram is waiting for something to set up? But that "something" is also waiting for log2ram?

azlux, 2022-01-04:
I just obersve a strange think into your output.
You have multiple tmpfs mount point on `/var/log` and `/var/hdd.log`. Only `/var/log` should have one (and only one)
Can you paste here the output of :
- `cat /etc/fstab`
- `cat /etc/log2ram.conf | grep PATH_DISK`

un99known99, 2022-01-04:
Thx for coming back @azlux :) and no worries, really appreciate ur support
here is the output

cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=738a4d67-01  /boot           vfat    defaults          0       2
PARTUUID=738a4d67-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

cat /etc/log2ram.conf | grep PATH_DISK
# example : PATH_DISK="/var/log;/home/test/FolderInRam"
PATH_DISK="/var/log"

xtvdata, 2022-01-04:
Just my 2cp.
As already said, if log2ram can be started without issues _after_ the boot it means that there is a dependency that is slowing it down for some reasons. Therefore it's all about finding what is causing the timeout.

In this view I'd suggest to check with `systemd-analyze`.

This command will generate a nice svg chart of the full systemd boot timings:
```bash
systemd-analyze plot >bootup.svg
```

And this will tell you the critical path to the _final_ target (it does not include the full dependency tree, but only the dependencies that are actually delaying the path):
```bash
systemd-analyze critical-chain
```
Also you can specify a unit to the last command, thus if you run `systemd-analyze critical-chain log2ram.service` you should get something like
```
The time when unit became active or started is printed after the "@" character.
The time the unit took to start is printed after the "+" character.

log2ram.service +1.931s
└─var-log.mount @7.353s
  └─dev-zram0.device @7.030s
```

And finally this one will produce a dot format output of the dependencies of _log2ram.service_:
```bash
systemd-analyze dot 'log2ram.service'
```
I think the output is already readable, but if you want it is possible to generate an SVG chart of it running the output through _dot_:
```bash
systemd-analyze dot 'log2ram.service' | dot -Tsvg >log2ram.svg
```
Note: to be able to render the chart you need the _graphviz_ package. That can be installed with `apt install graphviz`.

This should help in reducing the number of possible causes.

Ranbato, 2022-01-23:
I had what I thought was this same issue and found https://github.com/azlux/log2ram/issues/146 where log2ram errors if you try to use a directory off the root.  I had added _/tmp_ to my path list and that was causing the error.  Once I removed it the problem went away.  Adding the comment in the hope that it narrows down this fix a bit.

un99known99, 2022-02-03:
thx for the hint, but my /etc/log2ram.conf in that case looks clean:
PATH_DISK="/var/log"

jubie25, 2022-05-02:
+1 for me. For all of you: try to start a failed log2ram-boot via
`systemctl start log2ram`
and reboot. For me the very next boot is ok(!) but successive boots fail(!). So it seems that an installed log2ram does have timing/start sequencing problems and fails to start (look at start messages, it shows 
`A start job is configured for log2ram`
and counts up to 2min.)
Starting log2ram manually "repairs" things and things run as expected including next boot. Then something gets lost again...
This may meet observations where remove/reinstalls repair log2ram...
I've got two SVGs (faild and ok) but didn't find any hint to an error (but the root cause may be invisible for my knowledge).

jubie25, 2022-05-02:
I digged a little bit deeper and now I can confirm that the rsync call in `syncFromDisk()` does not come back. This causes the timeout.
No idea why this is so. cp is the offered alternative. What are the drawbacks to use cp instead?

eternalliving, 2023-06-15:
I would love to find a fix on this. I disabled log2ram and start it manually once it's all booted up. I run /var/mysql as a second directory to the default /var/log. However, I need to restart mariadb otherwise it writes into some kind of void and not to ram or the mysql.hdd/

When the power gets cut, it seems like I have half my database, but a few tables that resort back to when I set everything up before log2ram... this is probably the software that's writing to the database instead of log2ram, but it's behaviour is linked to having to start log2ram manually after the database is already loaded. 

How can we use more than 1 directory and have it successful on boot? I love this little piece of software and it's saved my SD card from a strange database rewrite bug that was writing 30+gb to a 32gb sd card daily!!

eternalliving, 2023-06-16:
Ok... so after looking further into it and following someone's suggestion with cp instead of rsync, and updating to the newest version (which I think is mainly the cp integration). 

Seems to be all good. Hopefully this will fix the strange database behaviour.