azlux/log2ram #178: Feature request

jliebich, 2021-11-22:
I use log2ram on different Raspberry Pi's.

The feature I miss, is a support for Telegram instead of E-Mail for the error system message if there is not enough place on RAM.

My proposal would be to make it similar as in this project: https://github.com/framps/raspiBackup

Proposal:

Add this options to the config file:

    # Telegram token
    DEFAULT_TELEGRAM_TOKEN=""
    # Telegram target chatid
    DEFAULT_TELEGRAM_CHATID=""

If these options are not empty, send the error system message if there is not enough place on RAM via Telegram.

If you do not know how to send Telegram messages via Bash script, you may refer to `function sendTelegramMessage()` in https://github.com/framps/raspiBackup/blob/master/raspiBackup.sh or https://thecodersblog.com/sending-the-message-on-telegram-using-cUrl/

azlux, 2022-03-04:
Hi,
I will not implement telegram or other message system (discord/teams/irc...) but we can add a custom command (in the config file) to use instead of the `mail`command.
This way you can use the command you want.
Good for you ?

xtvdata, 2022-03-04:
IMHO the best would be to just trigger a configurable user script to run in case of OOM in the RAM disk.
Something like:
```
ONERROR_USERSCRIPT="/path/to/my-script"
```

Providing a couple of arguments to the script call, could also be useful:
- a reason/error code (e.g. to determine if the issue is on `SIZE`, `LOG_DISK_SIZE` or _other_)
- the share (item of `PATH_DISK`) on which the issue occurred.

Then everyone can write his own script to either: send an email, send a message on Telegram, Pushover, etc...
This would also allow to option to integrate with more structured monitoring systems (e.g. Zabbix, etc...).

jliebich, 2022-03-04:
I like xtvdata's suggestion because it allows everyone (including me) to craft their own customized solution!

xtvdata, 2022-03-04:
On the other hand, it might be already possible to send messages leveraging other standard tools...

E.g.:
- https://www.rsyslog.com/doc/v8-stable/configuration/modules/ommail.html
- https://www.rsyslog.com/doc/v8-stable/configuration/modules/ompipe.html
- https://www.rsyslog.com/doc/v8-stable/configuration/modules/omprog.html

If you check rsyslog documentation and find an output module that fits your need it's a matter of writing a line of configuration for rsyslog.

The advantage of having an `ONERROR_USERSCRIPT` is that it would be easier to extrapolate details via arguments, however an error catch somewhere in log2ram with a standardized line in syslog (with a documented structure easy to parse), may achieve the same result (or even better - since it would be more standard and with less code to maintain for @azlux ).

azlux, 2024-06-08:
Add custom command here : https://github.com/azlux/log2ram/commit/d368260763935a9bae5d3ad2f16d0e30e127375b

closed 2024-09-19 by azlux