azlux/log2ram #184: Log2ram needs SIZE=4GB to run

Gifford47, 2022-03-03:
On my system log2ram needs about 4GB to run properly. Otherwise I the service doesn't start.
I have several programs running that are logging every x seconds.

Is it possible to get log2ram running with a log filesize smaller than 400mb?

xtvdata, 2022-03-03:
4GB?
Keeping 4GB of logs seems a bit extreme... ^_^

**Scenario A:**
_You are logging much more than is actually needed_
=> configure whatever is generating those huge logs to reduce verbosity to just what is really needed (e.g.: no debug)

**Scenario B:**
_There is something wrong in log rotation_
=> configure properly the logrotate (usually you need no more than 1-2 months of local logs)

**Scenario C:**
_You are a developer and you need debug_
=> limit only the history kept for the software in development to 1-2 weeks (via logrotate config)

**Scenario D:**
_You really need that amount of Logs_
=> in this case you probably need a dedicated log manager (on a separate dedicated server - not a virtual machine, nor a docker container) - consider a different tool to store logs, either on premise or in cloud (e.g.: [Loki](https://grafana.com/oss/loki/), [Elasticsearch](https://www.elastic.co), [Splunk](https://www.splunk.com), [GrayLog](https://www.graylog.org), and many other...). This will also allow you to be more compliant with privacy topics by setting up a proper retention rule.

Anyway, in a business production environment I'd still recommend to use a real disk to save logs locally, not a RAM disk (to keep log information in case of crash/network issues). If you are worried about FS consistency issues in case of crash, just keep the logs on a dedicated partition/disk (a separate disk unit will also distribute IOPS for heavy logging).

Gifford47, 2022-03-12:
Hi! Thanks for the answer!
I thing scenario a is right. I try to remove the unnecessary and orphaned logs.
But how is the min size log being calculated?

xtvdata, 2022-03-12:
Hi @Gifford47 ,

> But how is the min size log being calculated?

This is complicated topic, since is very dependent from what you are using.

This **first thing** to do is to double **check the log verbosity configuration of installed software** (a plain standard "server" installation generated a very limited amount of logs) and verify that indeed nothing is logging a debug level (of course, unless that it is the desired behavior). Set everything to `info`, `notice` or `warn`.
Which one should you choose?
That really depends on the verbosity and the contents logged in each level by each software, but do not set it above "warn", since in that case you will indeed loose very important information. 

**Note**: if you install from GitHub source be sure to use a _tagged_ release since master/main may have by default a very verbose development configuration.

The **second thing** to do is to **review the configuration of rsyslog** (assuming you are using Debian).
Rsyslog allows you to define custom rules (in `/etc/rsyslog.d/`) to.
- to cut some log lines before logs are actually saved on disk (e.g. for very verbose software)
- to split log files
- and many more actions (such as forward data to a log-server), but this is off-topic..

The **last thing** to do is to configure **log rotation** properly (`/etc/rsyslog.d/...`).
Log rotation comes with a standard configuration that works for a standard installation, however the software you install may not follow the rules and unless it provide a specific rotation rule, it may not be rotated properly.
Also depending on your needs, here you can define exactly how much history you want to keep (requirement that may vary for different softwares).

After this steps you should let the system run and watch the size of your `/var/log`, in the end set `log2ram` capacity to twice that amount (Ramdisk are dynamic thus if you don't use space ram will not be consumed).

**Note**: in log2ram there are 2 values: `SIZE` and `LOG_DISK_SIZE` and this can be tricky.
`SIZE` is the actual max size of the RAM disk, while `LOG_DISK_SIZE` is an additional limit (larger than `SIZE`) that is supposed to be defined taking into account the data compression ratio of the compression algorithm in use.
WARNING: while the table provided by @azlux in the documentation with the algorithm efficiency is accurate, it may vary (significantly) using special log formats/data. That reference table is valid for regular logs (plain text), but if you have some software that generates binary logs, or sparse files, the result could be very different and less predictable. (extreme, not realistic, case: if in `/var/log` the majority of the files would be JPEG images you'd end up with compression rate close to 1:1 ...)

closed 2023-03-13 by Gifford47