azlux/log2ram #186: Package 'log2ram' has no installation candidate

Codex-, 2022-03-07:
I was looking at installing log2ram and followed the steps to prepare `apt`, however, when performing the installation step I'm greeted with:
```
➜  log2ram-master sudo apt install log2ram                   
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package log2ram is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'log2ram' has no installation candidate
```

Fetch dump:
```
OS: Raspbian 10 buster
Kernel: armv7l Linux 5.10.63-v7l+
Uptime: 12h 24m
Packages: 629
Shell: 8357
CPU: ARMv7 rev 3 (v7l) @ 4x 1.5GHz [57.9°C]
GPU: 
RAM: 80MiB / 3660MiB
```

Not sure where to go from here, I could manually install but would like to help debug this issue if possible

azlux, 2022-03-08:
Hi,
What's the output of `cat /etc/apt/sources.list.d/azlux.list`?

Codex-, 2022-03-08:
```
➜  ~ cat /etc/apt/sources.list.d/azlux.list
deb [signed-by=/usr/share/keyrings/azlux-archive-keyring.gpg] http://packages.azlux.fr/debian/ bullseye main
```

ah I think I see why, using `buster` still, haven't looked into upgrading to `bullseye` yet

azlux, 2022-03-08:
Indeed

closed 2022-03-08 by azlux