azlux/log2ram #190: L2R doesn't save existing logs, programs not starting

gpmnsp, 2022-05-20:
I've tried both, manual and apt install.
After reboot there's only one file in /hdd.log: boot.log.
Worse: /nginx and /letsencrypt in /var/log have vanished. That prevents these programs from starting.
I re-created the respective dirs, nginx and letsencryt do work then.
But the very next day same situation. Both dirs are gone and also not in /hdd.log.
I have no clue.
What could possibly have gone wrong?

theruffus, 2022-06-19:
Same here... NGNIX broken, logs gone and log2ram won't start because there is no space in ram for hdd.log... My /var/log is like 3mb, and I set the log2ram size as high as 2gigs to no avail... this program is just broken or potentially incompatible with omv6.

gpmnsp, 2022-06-19:
Thx for reply, but this is'nt it.
I don't know or use OMV.
Mines just plain Raspbian OS. L2R does start, but not working as expected nor needed.

xtvdata, 2022-06-19:
Issues like these may happen when programs try to create a file or directory in an inexistent location (and the scenario is complicated by the rsync optimised behaviour).

You may try the following:
1. Stop log2ram (this should sync everything to the disk)
2. Create the eventually missing directories in /var/log
3. Restart log2ram (it should populate the ram-disc with all the contents of the disk copy of /var/log

If that is not enough, at point 2 try also to create a small file in each directory.

If everything is in place correctly it should survive a reboot.