azlux/log2ram #194: Ending up with non-ARM binaries

marcelstoer, 2022-08-08:
I am wondering, did anybody else around here experience this issue: https://raspberrypi.stackexchange.com/q/138610/116520?

azlux, 2022-08-09:
If you check the packages list directly on the repo https://packages.azlux.fr/debian/dists/buster/main/binary-armhf/Packages , you will see it don't hold any uptime/ps tool. so this repo cannot replace offlicial ones.

marcelstoer, 2022-08-09:
Yep, that was my point in the last comment over there at https://raspberrypi.stackexchange.com/questions/138610/system-binaries-fail-with-exec-format-error-apt-get-fails-with-117-structur#comment236395_138611. The user posting an "answer" pointed to this source list but I don't see the cause-effect relationship yet.

closed 2022-08-10 by marcelstoer