azlux/log2ram #202: rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]

jult, 2023-01-31:
This is on a plain up to date debian 11, as installed per your instructions. The funny thing is: There were no 'previous errors'.
Not that I can find anyway. A search in google tells me this has to do with dir permissions being wrong somewhere.

azlux, 2023-01-31:
It's look like you have ACL on your logs how cannot be write on the HDD mount point.
It's also possible to have this error if you have cut the copy in progress.

vincentstans, 2023-04-25:
I have the same issue on ubuntu 22.04 lts. dist-upgraded since ubuntu 16

error seems it could not do something with log2ram.log it was deleted or not created.

never the less it is up and running. I have created log2ram.log manually in /var/hdd.log and /var/log with sudo touch

Nuclear-Conger, 2023-05-04:
Same error on a fresh RaspberryPi OS Bullseye.

sudo touching those files made the error go away though (thank you @vincentstans).

daad919, 2023-07-12:
> It seems to be a bug in rsync 3.2.3, updating to 3.2.7 solved the issue for me.
> 
> https://linuxhint.com/update-rsync-raspberry-pi/

Would this also work on a promox server? I have a problem with some of my backups at the moment.

daad919, 2023-07-12:
Thanks,

Strange thing is it worked till 2 weeks ago, now i have problems with 2 of my containers...

daad919, 2023-07-12:
> It seems to be a bug in rsync 3.2.3, updating to 3.2.7 solved the issue for me.
> 
> https://linuxhint.com/update-rsync-raspberry-pi/

This worked also to update rsync on my proxmox setup. Thanks!

novastream, 2023-07-12:
Great, glad it worked :)

WirtsLegs, 2024-03-12:
same error on rsync 3.2.7-1 for me

zivagolee, 2024-07-06:
yea same issue with raspbian-bullseye. tried using rsync 3.2.7 and get the same errors on /var/log/journal files