azlux/log2ram #207: Has this been submitted to the Debian/Raspbian repositories? If not it should!

Danathar, 2023-06-17:
This is a great package. Just wondering why it's not in the distro's main repositories. It really should be.

Danathar, 2023-06-17:
Submitted a request to Debian to have this added to the wishlist for the main repository.

closed 2023-06-17 by Danathar

azlux, 2023-06-27:
Hi, I don't know where did you find my email, but it's not the good one, mine is : [github@azlux.fr](mailto:github@azlux.fr)
Thank for the ticket. 
Shouldn't be better to open it on raspbian repo instead of debian ?

Danathar, 2023-06-27:
I actually had to do some hunting to find your email ;) It was in the source license file. Regarding the Raspbian repository versus Debian, I figured that since Raspbian is based on Debian, having the issue in Debian's repository would cover both, as Raspbian would pick it up once it's in Debian's repos.

Danathar, 2023-06-27:
Just sent an update to the Debian Bug database for the ticket to have the email changed.