azlux/log2ram #211: Log2Ram won't start on RPi OS 64-bit

AleXSR700, 2023-07-26:
Hi everyone,

after performing `sudo journalctl --vacuum-size=32M` I still cannot get log2ram to start.

Can anybody see what the issue is? Maybe it is something really simple, but I just don't understand it :(

```
piserver@piserver:~ $ sudo du -h /var/log/ | tail -n1
106M	/var/log/
piserver@piserver:~ $ sudo perl -pe 's/^MAIL=.*$/MAIL=false/g' -i /etc/log2ram.conf
piserver@piserver:~ $ sudo log2ram start
ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync.
piserver@piserver:~ $ sudo du -hs /var/log/* | sort -h | tail -n 3
292K	/var/log/daemon.log
492K	/var/log/syslog
105M	/var/log/journal
piserver@piserver:~ $ sudo journalctl --vacuum-size=32M
Vacuuming done, freed 0B of archived journals from /run/log/journal.
Vacuuming done, freed 0B of archived journals from /var/log/journal.
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@485f0c6774204f52b01570fc8af3ce8a-0000000000000001-0005f0695944e9e3.journal (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@0005f0695948d162-247631b0f3186384.journal~ (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@485f0c6774204f52b01570fc8af3ce8a-00000000000001de-0005f0695974c178.journal (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@1abfcd8e9f4b4775bafe5fc0751828a6-0000000000000001-0005f069597f29f6.journal (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@0005f069598024c4-a542f995f9ab099a.journal~ (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@0005fac195547afd-049279486836c4a8.journal~ (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/user-1000@ee63c7469b0b4946b1ab728a73b5ffeb-0000000000000267-0005fac195000f4f.journal (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@9f0668c7f1034b1783d23f92ac5999d2-0000000000000001-0005fac1955071f4.journal (8.0M).
Deleted archived journal /var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@00060166bcf1cccc-5c919929d63b5a1f.journal~ (8.0M).
Vacuuming done, freed 72.0M of archived journals from /var/log/journal/0c7beb57739d45e08297f4a37e361c8f.
piserver@piserver:~ $ sudo log2ram start
building file list ... done
Xorg.0.log
Xorg.0.log.old
alternatives.log
auth.log
boot.log
bootstrap.log
btmp
daemon.log
debug
dpkg.log
faillog
fontconfig.log
kern.log
lastlog
messages
syslog
user.log
vncserver-x11.log
vncserver-x11.log.bak
wtmp
apt/
apt/eipp.log.xz
apt/history.log
apt/term.log
cups/
cups/access_log
cups/error_log
hp/
hp/tmp/
journal/
journal/0c7beb57739d45e08297f4a37e361c8f/
journal/0c7beb57739d45e08297f4a37e361c8f/system.journal
journal/0c7beb57739d45e08297f4a37e361c8f/system@00060166f32efd33-43289ca19270a100.journal~
rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system.journal","user.crtime_usec") failed: Operation not supported (95)
journal/0c7beb57739d45e08297f4a37e361c8f/user-1000@a666415c3353417b8f0fec8d91d2b625-00000000000002f3-0006016829e7e7eb.journal
rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0c7beb57739d45e08297f4a37e361c8f/system@00060166f32efd33-43289ca19270a100.journal~","user.crtime_usec") failed: Operation not supported (95)
journal/0c7beb57739d45e08297f4a37e361c8f/user-1000@ee63c7469b0b4946b1ab728a73b5ffeb-00000000000003d6-00060166b9c80217.journal
rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0c7beb57739d45e08297f4a37e361c8f/user-1000@a666415c3353417b8f0fec8d91d2b625-00000000000002f3-0006016829e7e7eb.journal","user.crtime_usec") failed: Operation not supported (95)
lightdm/
lightdm/lightdm.log
lightdm/lightdm.log.old
lightdm/x-0.log
lightdm/x-0.log.old
private/
runit/
runit/ssh/
rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0c7beb57739d45e08297f4a37e361c8f/user-1000@ee63c7469b0b4946b1ab728a73b5ffeb-00000000000003d6-00060166b9c80217.journal","user.crtime_usec") failed: Operation not supported (95)
deleting log2ram.log

sent 35,496,103 bytes  received 672 bytes  23,664,516.67 bytes/sec
total size is 35,484,476  speedup is 1.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]
piserver@piserver:~ $ sudo systemctl log2ram status
Unknown command verb log2ram.
piserver@piserver:~ $ sudo systemctl status log2ram
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-12-22 12:55:47 CET; 7 months 3 days ago
    Process: 361 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
   Main PID: 361 (code=exited, status=1/FAILURE)
        CPU: 128ms

Warning: journal has been rotated since unit was started, output may be incomplete.
piserver@piserver:~ $
```

sysadmiral, 2023-10-15:
It works for me if I set the `--vacuum-size` to something more aggresive:

`sudo journalctl --vacuum-size=10M`

peterpepo, 2023-10-15:
@sysadmiral you're right, I had a same problem and same solution.
The thing is, journalctl isn't the only one adding entries in /var/log.
Total size of /var/log can be retrieved by
```
sudo du -sh /var/log
23M     /var/log
```
This puzzled me for a while, since I had `SystemMaxUse=20M` in my `/etc/systemd/journald.conf` and we clearly see, that /var/log is already bigger than that.  (In my `/etc/log2ram.conf` I had `SIZE=40M`) which was still causing issues, because my /var/log was bigger than that couple days ago. Since then I vacuumed to 0.1s

As a proof - list of entries, sorted by largest can be found running command from https://github.com/azlux/log2ram#existing-content-in-varlog-too-large-for-ram
In my case:
```
sudo du -hs /var/log/* | sort -h | tail -n 10
344K    /var/log/syslog.2.gz
352K    /var/log/wtmp
388K    /var/log/auth.log.1
532K    /var/log/daemon.log
744K    /var/log/syslog
1.8M    /var/log/kern.log.1
1.8M    /var/log/messages.1
2.6M    /var/log/journal
5.1M    /var/log/daemon.log.1
7.0M    /var/log/syslog.1
```

At the moment I set following, since I'm already running zram:
```
SIZE=64M
ZL2R=true
LOG_DISK_SIZE=128M
```

This means, if I understand it correctly, that I effectively have up to 128MB for /var/log which gets compressed to zram where it occupies up to 64MB.

I also reduced `SystemMaxUse=10M` in my `/etc/systemd/journald.conf` as I don't care about logs.

This will hopefully suffice until I upgrade for nuc with proper ssd and expandable ram.