azlux/log2ram #214: Changeing timer does not work

spacemanspiff2007, 2023-10-20:
I tried changing the sync interval to weekly following the description:
```
sudo systemctl edit log2ram-daily.timer
```
```
### Editing /etc/systemd/system/log2ram-daily.timer.d/override.conf
### Anything between here and the comment below will become the new contents of the file

[Unit]
Description=Daily Log2Ram writing activities

[Timer]
OnCalendar=Sun *-*-* 3:55:00
Persistent=true

[Install]
WantedBy=timers.target

### Lines below this comment will be discarded

### /etc/systemd/system/log2ram-daily.timer
# [Unit]
# Description=Daily Log2Ram writing activities
# 
# [Timer]
# OnCalendar=*-*-* 23:55:00
# Persistent=true
# 
# [Install]
# WantedBy=timers.target
```
```
systemctl list-timers
```
```
Fri 2023-10-20 23:55:00 CEST 18h left       Fri 2023-10-20 05:35:03 CEST 15min ago         log2ram-daily.timer          log2ram-daily.service
```

Somehow the timer does not get picked up.
What am I doing wrong?

peterpepo, 2023-10-25:
Please add `OnCalendar=` above your `OnCalendar=Sun *-*-* 3:55:00` as instructed in README - https://github.com/azlux/log2ram#refresh-time.

As explained here, https://bbs.archlinux.org/viewtopic.php?pid=1642051#p1642051

> Timers are cumulative, so in the override you need to reset the existing timers otherwise your OnCalendar setting gets added to the existing one.

I have tested this on my machine

`
pi@raspberrypi:~ $ sudo systemctl list-timers --all
NEXT                         LEFT          LAST                         PASSED       UNIT                         ACTIVATES
Sun 2023-10-29 03:55:00 CET  3 days left   Tue 2023-10-24 23:55:02 CEST 9h ago       log2ram-daily.timer          log2ram-daily.service
`

spacemanspiff2007, 2023-10-29:
Thank you for your help - it works now as expected

closed 2023-10-29 by spacemanspiff2007