azlux/log2ram #218: Latest APT version (1.6.1) far behind Git repo

creeper6530, 2023-11-25:
Please, would you release a new version with all the improvements from the Git version? I'd especially like to have journald awareness, but I can't install anything other than via APT

jult, 2024-01-10:
Yes, please. Another vote to up the apt for debian bookworm.

onlineapps-cloud, 2024-02-27:
i added my vote

BugBuster1701, 2024-03-24:
Yes, please release a new version.

adamsthws, 2024-06-06:
I would like to kindly echo this request too.

azlux, 2024-06-06:
Hi all,

Since people request for a new release, I need to adapt the project state to the new need.
Since I'm busy with other stuff, I do less computer/coding. And I don't have raspberry with sd cart anymore (only ssd) with log2ram.

What I propose, I will make a beta release for log2ram, and propose a vote ? / feedback ? on it.
If (let's say) five peoples approve the beta, I will merge to stable one week later.

What do you think ?
I want to avoid a new [xz](https://www.docker.com/blog/openssh-and-xz-liblzma/) problem.

Az

adamsthws, 2024-06-07:
Thankyou Az, that would be super!I'd be happy to undertake testing. 
Is there a changelog so we know where to focus the testing efforts?

azlux, 2024-06-08:
Let's test the new version with https://github.com/azlux/log2ram/releases/tag/1.7b1 available in testing (sid) codename

adamsthws, 2024-06-09:
> Let's test the new version with https://github.com/azlux/log2ram/releases/tag/1.7b1 available in testing (sid) codename

I'm happy to report no major issues!

The only minor issue I found was that the Ansible playbook needed updating:
[(Pull Request 225)](https://github.com/azlux/log2ram/pull/225)

Other than that, all working well for me :)

adamsthws, 2024-06-10:
Hi Az, 

With the new beta needing testers, do you think it might be a good idea to make a post on Reddit (r/raspberry_pi?) requesting as such?

I'd be happy to compose the Reddit post. Though I understand you're currently on holiday so maybe not this week.

Cheers,
Adam

closed 2024-06-28 by azlux