azlux/log2ram #219: log2ram failed to update writes to disk at reboot

jult, 2024-01-10:
I have seen this a couple of times on different debian bookworm instances where log2ram is installed;
When I do a reboot (shutdown -r now, or reboot), I see in syslog that the log2ram service indeed stops, but after having returned and checking the hdd.* dir contents, they are not updated to the latest versions that were there, and the current version at that boot is an older one. Pretty bad actually. I've lost some work due to not expecting this.
Something's not correctly running rsync back to disk properly around shutting down. I have not been able to pinpoint the cause.

I have installed log2ram via apt, and not changed service timers or anything, only the dirs that should be in RAM.

p1r473, 2024-04-13:
I'm having a similar problem 
If I do `logger mytest` and then `log2ram write` then `reboot` and then `cat /var/log/syslog | grep mytest` I can't find my log entry after reboot

scara, 2024-04-15:
> Which distro ? I think it's mostly an issue with ubuntu. It works fine on debian.

Thanks for your reply: there is a more specific follow-up here: #222.

HTH,
Matteo

bondskin, 2024-07-20:
same here on Raspberry Debian Bookworm

ioogithub, 2024-09-14:
@jult  can you look in /var/journal/ after a reboot and see if you have any files with ~ at the end of their name indicating a corrupted journal?