azlux/log2ram #221: failed to update host

fiergna, 2024-03-23:
hi, 

trying to update my host, but because the http://packages.azlux.fr/debian does not resolve / have a Release file, i can't do anything.
is this pakage still active / mantained?

`sudo apt update && sudo apt upgrade -y
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:3 https://packages.wazuh.com/4.x/apt stable InRelease
Ign:4 http://packages.azlux.fr/debian bookworm InRelease
Err:5 http://packages.azlux.fr/debian bookworm Release
  404  Not Found [IP: 2a01:728:401:1c::100 80]
Reading package lists... Done
E: The repository 'http://packages.azlux.fr/debian bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

BugBuster1701, 2024-03-23:
According to the website, the packages are now available at `https://azlux.fr/debian/`, but the key is no longer loadable, so it is no longer usable when you set it up for the first time.

hadfiiw, 2024-03-23:
Is this really the same as 220? The public key not being available (what BugBuster1701 referenced) is new as of yesterday while issue 220 has been open for a month.

JakobLichterfeld, 2024-03-23:
> Is this really the same as 220? The public key not being available (what BugBuster1701 referenced) is new as of yesterday while issue 220 has been open for a month.

Both have the issue of the repo move, bot fail with apt imo

hadfiiw, 2024-03-23:
Not sure what changed but using the original URL now works again.

```
echo "deb [signed-by=/usr/share/keyrings/azlux-archive-keyring.gpg] http://packages.azlux.fr/debian/ bookworm main" | sudo tee /etc/apt/sources.list.d/azlux.list
sudo wget -O /usr/share/keyrings/azlux-archive-keyring.gpg  https://azlux.fr/repo.gpg
sudo apt update
sudo apt install log2ram
```

Saudumm, 2024-03-23:
I can confirm that it works again.

fiergna, 2024-03-24:
can also confirm that it works again, i had to add the source again because the GPG key changed

closed 2024-03-24 by fiergna

azlux, 2024-03-24:
Hi,
I wasn't able to fix it during the day because of my holiday. So it's was down a full day.
Nothing have change repo/key are exactly the same.

Best regards,
Azlux