azlux/log2ram #222: Logs not writing at reboot?

p1r473, 2024-04-13:
If I do 
```
logger mytest
log2ram write
reboot
cat /var/log/syslog | grep mytest
```
I can't find my log entry after reboot

Anyone have this issue?

scara, 2024-04-14:
Hi @p1r473,
are you sure that it will be written into the `syslog` file?
I'd give `user.log` a try (CEST time here):
```
$ logger mytest
$ sudo log2ram write
$ cat /var/log/user.log
Apr 14 22:02:37 pi401 pi: mytest
```
HTH,
Matteo

p1r473, 2024-04-14:
Ive checked
/var/log/user.log
/var/log/syslog
/var/hdd.log/user.log
/var/hdd.log/syslog

But whatever I write into /var/log isnt persisting reboot
Every time I reboot, my logs reset to what they were when I first installed log2ram

Before reboot:
```
❯ logger mytest; logger log2ram write                                                                                                                     
❯ cat /var/hdd.log/user.log | grep mytest                                                                                                                 
❯ cat /var/hdd.log/syslog | grep mytest                                                                                                                  
❯ cat /var/log/syslog | grep mytest
Apr 14 16:25:36 Monkeebutt root: mytest                                                                                                                                                                                                                              
❯ cat /var/log/user.log | grep mytest
Apr 14 16:25:36 Monkeebutt root: mytest         
```

After reboot:
```
cat /var/hdd.log/user.log | grep mytest                                                                                                                 
❯ cat /var/hdd.log/syslog | grep mytest                                                                                                                  
❯ cat /var/log/syslog | grep mytest                                                                                                                      
❯ cat /var/log/user.log | grep mytest     
```

My config:
`/etc/log2ram.conf`
```
SIZE=500M
USE_RSYNC=false
MAIL=true
PATH_DISK="/var/log"
ZL2R=false
COMP_ALG=lz4
LOG_DISK_SIZE=500M  
```
` /etc/systemd/system/log2ram.service.d/override.conf`
```
[Service]
ExecStart= /usr/local/bin/log2ram start --timeout=900     
```
` /etc/systemd/system/log2ram-daily.timer.d/override.conf`
```
[Timer]
OnCalendar=
OnCalendar=*-*-* *:00/6 
```

p1r473, 2024-04-14:
Ive checked
/var/log/user.log
/var/log/syslog
/var/hdd.log/user.log
/var/hdd.log/syslog

But whatever I write into /var/log isnt persisting reboot
Every time I reboot, my logs reset to what they were when I first installed log2ram

Before reboot:
````
logger mytest; log2ram write   
building file list ... done
./
syslog                                                                                             
❯ cat /var/hdd.log/user.log | grep mytest
Apr 14 16:34:39 Monkeebutt root: mytest                                                                                                                   
❯ cat /var/hdd.log/syslog | grep mytest
Apr 14 16:34:39 Monkeebutt root: mytest                                                                                                                  
❯ cat /var/log/syslog | grep mytest
Apr 14 16:34:39 Monkeebutt root: mytest                                                                                                                  
❯ cat /var/log/user.log | grep mytest
Apr 14 16:34:39 Monkeebutt root: mytest            
```

After reboot:
```
❯ cat /var/hdd.log/user.log | grep mytest                                                                                                                 
❯ cat /var/hdd.log/syslog | grep mytest                                                                                                                  
❯ cat /var/log/syslog | grep mytest                                                                                                                      
❯ cat /var/log/user.log | grep mytest     
```

My config:
`/etc/log2ram.conf`
```
SIZE=500M
USE_RSYNC=false
MAIL=true
PATH_DISK="/var/log"
ZL2R=false
COMP_ALG=lz4
LOG_DISK_SIZE=500M  
```
` /etc/systemd/system/log2ram.service.d/override.conf`
```
[Service]
ExecStart= /usr/local/bin/log2ram start --timeout=900     
```
` /etc/systemd/system/log2ram-daily.timer.d/override.conf`
```
[Timer]
OnCalendar=
OnCalendar=*-*-* *:00/6 
```

p1r473, 2024-04-15:
I think its failing to copy from /var/hdd.log back to /var/log ?

closed 2024-05-01 by p1r473

ioogithub, 2024-09-14:
@p1r473 can you look in /var/journal/ after a reboot and see if you have any files with ~ at the end of their name indicating a corrupted journal?