azlux/log2ram #233: All journal logs always lost on reboot on ubuntu 24.04

ioogithub, 2024-09-14:
The app does not work for ubuntu 24.04.1
 Tested on a clean system with very little software and stock configurations.

Journal logs are not saved and restored after a system reboot.

To test:

1. Clear the journal with journal --rotate and journalctl --vacuum-time=1s.
2. Reboot
3. Run journalctl --list-boots and observe 1 boot entry is recorded.
4. Reboot
5. Run journalctl--list-boots and observe only the last 1 entry is recorded. All journal logs before the most recent boot are lost.
6. Double chech: Run journalctl and observe the log entries start only from the most recent boot.

Journalctl -t log2ram showes rsync copy records from after the boot including /journal/* files but nothing before the boot as those logs are lost.

Is this by design? I can't see log2ram being useful if it wipes out the logs after every reboot. The only change to log2ram was to increase size to 2560M. 

It looks like this could be a bug because the log2ram.cfg that was installed has double entries for the JOURNAL_AWARE=true setting:
```
# Should log2ram consider journald and do a log rotate before copying the log files back? Please note that for this
# rsync is mandatory. Also make sure that you have configured SystemMaxUse in journald.conf, because the size of this
# ram folder is will probably not be as large as journald will use by default
JOURNALD_AWARE=true

# Set to 'true' to enable log rotation for journald logs before syncing. 
# Note: 'rsync' must be used for this feature. 
# Ensure 'SystemMaxUse' is configured in 'journald.conf' 
#  - (To limit journald’s disk usage to be smaller than the log2ram RAM allocation).
```
Was this part of the app modified recently? If so it does not work.

ioogithub, 2024-09-14:
Tested a few more times, removed --purged log2ram and reinstalled.

Here is what is happening. After a boot log2ram saves the journal entries from the first boot before it was installed and the last previous boot only. All other journal entries are deleted.

So if I boot 5 times 8n journalctl I would see -1 and 0:

-1 = Boot 1
0 = Boot 5 (the most recent boot).

All entries for boot 2, 3, and 4 are missing.

I can easily troubleshoot what log2ram is doing before the boot because of course all the logs are gone.

Definately not working as expected.

ioogithub, 2024-09-14:
I am not sure but I think this could be a similar issue to: https://github.com/azlux/log2ram/issues/219

ioogithub, 2024-09-14:
Looking deeper into this it appears that in the /var/log/journal directory the logs are still there but logs from the previous boot are corrupted with a ~ after the filename. So it looks like the problem is indeed this line:

JOURNALD_AWARE=true

log2ram is corrupting the journal when it tries to rotate.

I think it is a similar issue to this which was marked as completed but I think the issue has not been fixed:
https://github.com/azlux/log2ram/issues/185

ioogithub, 2024-09-14:
Discussion of the specific issue is here: https://github.com/azlux/log2ram/pull/200

azlux, 2024-09-18:
The double `JOURNAL_AWARE` don't make any issue.
Since it's minor, I didn't release any new version to fix that even if it's already fix into master branch.

The option `JOURNALD_AWARE`  allwo log2ram to make a journalctl rotation befor sync the /var/log folder : https://github.com/azlux/log2ram/blob/master/log2ram#L29
The code force rsync to include only journals with a `@` : https://github.com/azlux/log2ram/blob/master/log2ram#L44

Let's try to investigate in this ticket instead of spreading into multiple issue.

azlux, 2024-09-20:
Can you check also the last commit https://github.com/azlux/log2ram/commit/83774d470ba9ed1d2ab286a390fee65b28272d34 , it's maybe link to your issue.

XhmikosR, 2025-09-02:
Unfortunately, this still happens for me on Raspberry Pi OS 12 64-bit lite too. The weird thing is that I don't hit the issue on a Raspberry Pi Zero 2 W, but only on my Raspberry Pi 4.

@azlux here's the analyze blame report you asked above in case it helps

[image]

XhmikosR, 2025-12-14:
Coming back at this, it seems the cause is that I hadn't set a `SystemMaxUse` for journal. With this set I consistently get everything right. I have also explicitly switched `Storage` to volatile.

```console
pi@rpi5:~ $  cat /etc/systemd/journald.conf.d/80-raspi-config-journal-storage.conf
# Created/managed by raspi-config

pi@rpi5:~ $  cat /etc/systemd/journald.conf | grep SystemMaxUse
SystemMaxUse=100M

pi@rpi5:~ $  sudo journalctl --list-boots
IDX BOOT ID                          FIRST ENTRY                 LAST ENTRY
-10 d6a41716912d46f48a7fd39b7bf034f2 Tue 2025-11-04 00:45:00 EET Sun 2025-11-09 07:42:45 EET
 -9 10d4651af19448be9ba6c203c9695db4 Sun 2025-11-09 07:43:01 EET Tue 2025-11-11 17:01:10 EET
 -8 f45ef4aa12964ef9933f7fce6e3ef6b9 Tue 2025-11-11 17:01:26 EET Fri 2025-11-14 08:02:21 EET
 -7 924baf836ba649bdb1a042722a253c06 Fri 2025-11-14 08:02:38 EET Sat 2025-11-15 23:08:44 EET
 -6 33565645020b4ea496b95880d3064fcf Sat 2025-11-15 23:09:01 EET Mon 2025-11-17 09:33:30 EET
 -5 4842e9456d0e42c980c378dace75ec71 Mon 2025-11-17 09:33:47 EET Fri 2025-12-05 08:05:20 EET
 -4 575ca46063a94f8eaa34efe9ce7c0062 Fri 2025-12-05 08:05:37 EET Sun 2025-12-07 20:01:53 EET
 -3 268136aea06342629475e2749b79ab20 Sun 2025-12-07 20:01:57 EET Tue 2025-12-09 17:37:09 EET
 -2 633d8e8c1a6c4784921fc209e25d999e Tue 2025-12-09 17:37:27 EET Sat 2025-12-13 18:42:00 EET
 -1 2e49e4f48c16486c8fc3ea17c8392f74 Sat 2025-12-13 18:42:18 EET Sun 2025-12-14 18:14:57 EET
  0 1c7d842fae964ccb9233fd0bb268e725 Sun 2025-12-14 18:15:15 EET Sun 2025-12-14 23:01:14 EET
```