azlux/log2ram #234: Journal files missing the + ACL

moodeaudio, 2024-09-19:
Hi,

Testing the super nice log2ram on our RaspiOS Bookworm build and I noticed some potential permission issues.

__Permission warning__
```
pi@trx:~ $ systemctl status log2ram
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; preset: enabled)
     Active: active (exited) since Thu 2024-09-19 17:09:16 EDT; 38min ago
   Main PID: 334 (code=exited, status=0/SUCCESS)
        CPU: 89ms

Warning: some journal files were not opened due to insufficient permissions.
```
__Missing + ACL on journal log files__
```
pi@trx:~ $ ls -l /var/log/journal/b5924457f74f4eb491585dd96cee36e8/
total 7072
-rw-r-----  1 root systemd-journal 1029832 Sep 19 17:09 system@ed446478f50049e1b812a76cf3695410-000000000009b23c-00062278a3eccdc4.journal
-rw-r-----  1 root systemd-journal  673192 Sep 19 17:09 system@ed446478f50049e1b812a76cf3695410-000000000009b989-0006227f134c89c7.journal
-rw-r-----+ 1 root systemd-journal 2621440 Sep 19 17:48 system.journal
-rw-r-----  1 root systemd-journal  120416 Sep 19 17:09 user-1000@1e203e7be1fb4ac2b828bba44ca0182a-000000000009bb31-0006227f202219c8.journal
-rw-r-----  1 root systemd-journal  136552 Sep 19 17:09 user-1000@7a272b04031346dc8756a7b7ed6ea301-000000000009b26b-000622796c1f0b22.journal
-rw-r-----+ 1 root systemd-journal 2621440 Sep 19 17:47 user-1000.journal
```
With log2ram off the + ACL's are present on all the files.

-Tim

azlux, 2024-09-19:
Can you try to add the option `-A`to rsync on the two lines : https://github.com/azlux/log2ram/blob/master/log2ram#L49 and https://github.com/azlux/log2ram/blob/master/log2ram#L78

From the rsync doc : 
>  --acls, -A               preserve ACLs

I didn't knew journactl was using ACL on their files

moodeaudio, 2024-09-19:
Adding -A is an improvement. 
`rsync -AaXv `

There are still some missing + but also a lot more new files.

```
pi@moode9:~ $ systemctl status log2ram
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; preset: enabled)
     Active: active (exited) since Thu 2024-09-19 18:15:39 EDT; 34s ago
   Main PID: 356 (code=exited, status=0/SUCCESS)
        CPU: 114ms

Sep 19 18:15:39 moode9 log2ram[510]: sysstat/
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sa16
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sa17
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sa18
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sa19
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sar16
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sar17
Sep 19 18:15:39 moode9 log2ram[510]: sysstat/sar18
Sep 19 18:15:39 moode9 log2ram[510]: sent 17,798,191 bytes  received 1,587 bytes  35,599,556.00 bytes/sec
Sep 19 18:15:39 moode9 log2ram[510]: total size is 17,786,118  speedup is 1.00

pi@moode9:~ $ ls -l /var/log/journal/86cd226206f445caa9798a72a76dd19b/
total 15616
-rw-r-----  1 root systemd-journal 1026688 Sep 19 14:35 system@52c5d98641a848a8a8e53016f24be2ab-0000000000003aaa-00062277607c716e.journal
-rw-r-----  1 root systemd-journal 1094160 Sep 19 14:54 system@52c5d98641a848a8a8e53016f24be2ab-0000000000004019-0006227d3037706e.journal
-rw-r-----  1 root systemd-journal  659312 Sep 19 08:32 system@52c5d98641a848a8a8e53016f24be2ab-00000000000040f5-00062278104a01f2.journal
-rw-r-----  1 root systemd-journal  905968 Sep 19 14:35 system@52c5d98641a848a8a8e53016f24be2ab-0000000000004105-0006227892967546.journal
-rw-r-----  1 root systemd-journal  695560 Sep 19 08:59 system@52c5d98641a848a8a8e53016f24be2ab-0000000000004106-000622781da59516.journal
-rw-r-----  1 root systemd-journal  544920 Sep 19 09:04 system@52c5d98641a848a8a8e53016f24be2ab-0000000000004108-000622787f549d3e.journal
-rw-r-----  1 root systemd-journal 1023208 Sep 19 14:55 system@52c5d98641a848a8a8e53016f24be2ab-000000000000465a-0006227d736494cf.journal
-rw-r-----+ 1 root systemd-journal 1110616 Sep 19 18:15 system@52c5d98641a848a8a8e53016f24be2ab-0000000000004bc4-0006227d78d0b6e5.journal
-rw-r-----  1 root systemd-journal  758224 Sep 19 16:57 system@52c5d98641a848a8a8e53016f24be2ab-0000000000005258-0006227d843b75ca.journal
-rw-r-----  1 root systemd-journal  679880 Sep 19 17:06 system@52c5d98641a848a8a8e53016f24be2ab-000000000000525d-0006227f2c3659bc.journal
-rw-r-----+ 1 root systemd-journal  710968 Sep 19 18:15 system@52c5d98641a848a8a8e53016f24be2ab-0000000000005264-0006227f4be999e3.journal
-rw-r-----+ 1 root systemd-journal 2621440 Sep 19 18:16 system.journal
-rw-r-----  1 root systemd-journal  117928 Sep 19 14:35 user-1000@03b4bbb252cf407386ad04e97c4e8eb9-0000000000003cd6-000622776820e299.journal
-rw-r-----  1 root systemd-journal  128880 Sep 19 14:35 user-1000@11bf274532b14b258d30656f9cb4efbb-00000000000042d3-00062278cc1c52ce.journal
-rw-r-----  1 root systemd-journal  120496 Sep 19 14:54 user-1000@1eb2adda2a5e4d4eabacd9abbccc5e3b-000000000000425e-0006227d32a95a58.journal
-rw-r-----  1 root systemd-journal  117152 Sep 19 08:32 user-1000@3db7120786254b8090974d3b9869ace8-0000000000004227-00062278114d4de5.journal
-rw-r-----  1 root systemd-journal  118112 Sep 19 09:04 user-1000@4fdf7337ab0847c89d8803d49e283a27-0000000000004241-000622788d20631a.journal
-rw-r-----+ 1 root systemd-journal  114968 Sep 19 18:15 user-1000@6e0cd391fc1640c09fb163dcbd1b2d07-0000000000004e15-0006227d80566d43.journal
-rw-r-----  1 root systemd-journal  117968 Sep 19 14:55 user-1000@6eef9702df6647efa639dd3fb90b9b40-0000000000004864-0006227d74d31471.journal
-rw-r-----+ 1 root systemd-journal  117152 Sep 19 18:15 user-1000@755299fda5e04701993b324b05f77fe5-0000000000005419-0006227f4e1b0b92.journal
-rw-r-----  1 root systemd-journal  119240 Sep 19 17:06 user-1000@8596f874b3e149cca48b180b5f4dc3d1-0000000000005419-0006227f419b9d98.journal
-rw-r-----  1 root systemd-journal  129728 Sep 19 16:57 user-1000@de491ceec19e4c049b7fe60416cfcbab-00000000000053f2-0006227d85adc227.journal
-rw-r-----  1 root systemd-journal  119648 Sep 19 08:59 user-1000@e6d6807a9290475c8908e0b48fc46441-0000000000004233-000622781ea80ec9.journal
-rw-r-----+ 1 root systemd-journal 2621440 Sep 19 18:16 user-1000.journal
pi@moode9:~ $ 
```

azlux, 2024-09-19:
All journal files not containing a "`@`" is normal, it's expected to avoid journal corruption.
Is there are others files missing on the copy  ?

moodeaudio, 2024-09-19:
I'm not too familiar with how log2ram works so can't be sure about the files but what I posted was just after a reboot.

azlux, 2024-09-19:
All good for me, I will push an update into log2ram

moodeaudio, 2024-09-19:
Fantastic, thanks :-)

commit 83774d47 (2024-09-20): Fix #234

Update rsync parameter : 
Adding -A to copy ACL as well
closed 2024-09-19 by moodeaudio