azlux/log2ram #240: [Feature Request]

camphamp, 2025-01-23:
Noting:
(1) [qwinz issue #173](https://github.com/azlux/log2ram/issues/173#issue-1000374639) with SD wear and tear.
(2) I very rarely care about log entries from before my last boot.

I would like to experiment starting with an empty /var/log at every boot and **no longer syncing the /var/log directory to disk**.

It would be ideal to have a configurable item in your config to toggle sync ON/OFF.

Is there an easy or recommended way to do this outside of your system (like a symlink of log.hdd -> /dev/null or commenting out the rsync commands (I cannot even find where they live)?

Thanks for your tool and you thoughts and consideration for this feature!

Sailing74, 2025-01-23:
What would you be missing if you simply setup /var/log as a RAM disk, just an entry in fstab?

camphamp, 2025-01-23:
It would provide the admin with an option to toggle log persistence "ON", for those cases when they need to troubleshoot a recurring crash event or something that only comes up over longer periods (weeks/months).

As mentioned, this is rare occasion for me, but if I need to collect logs between boots then I'd use this tool just for that facility.

bughunter2, 2025-07-20:
> What would you be missing if you simply setup /var/log as a RAM disk, just an entry in fstab?

Setting up a /var/log entry as a RAM disk is not enough, because the RAM disk should be initialized with a copy of the existing log files in order to preserve directory structures that were created by package scripts, such as **exim4** and **nginx**. Otherwise, such services will fail to start. Hence, something like log2ram is very useful for this purpose.

And that's why it would be a very useful feature to allow disabling rsync completely for people who like to use log2ram but prevent ANY kind of wear and tear on their storage devices.