azlux/log2ram #241: log2ram doesn't play nice with zramswap (from zram-tools package)

reukiodo, 2025-02-20:
It seems that both log2ram and zramswap (from the zram-tools package) want to use /dev/zram0, but log2ram service starts earlier in boot and claims it first, so after installing log2ram, zramswap service always fails.

reukiodo, 2025-02-20:
upon much deeper inspection into `/usr/local/bin/log2ram` I see:
```
create_zram_log_drive() {
    # Check Zram Class created
    if [ ! -d "/sys/class/zram-control" ]; then
        modprobe zram
        RAM_DEV='0'
    else
        RAM_DEV=$(cat /sys/class/zram-control/hot_add)
    fi
    echo "$COMP_ALG" >"/sys/block/zram${RAM_DEV}/comp_algorithm"
    echo "$LOG_DISK_SIZE" >"/sys/block/zram${RAM_DEV}/disksize"
    echo "$SIZE" >"/sys/block/zram${RAM_DEV}/mem_limit"
    mke2fs -t ext4 "/dev/zram${RAM_DEV}"
}
```
where in `/usr/sbin/zramswap` there is only:
```
readonly SWAP_DEV="/dev/zram0"
```
this seems to me that log2ram uses the first (zram0) when zram module hasn't been loaded (which is always the case at startup), and creates new zram devices afterwards (restarts?), where zramswap (which loads later at boot) seems to always assume that zram module has never been loaded and claims the first device - is this accurate?

if this is true, then maybe the real issue is in the zram-tools package, but that would need a change in the debian package

reukiodo, 2025-02-21:
in this case, I 'fixed' it by changing the code in `/usr/sbin/zramswap` to https://github.com/reukiodo/zram-tools/blob/master/zramswap.sh

andbad, 2026-01-29:
I had the same problem with zram-config (Ubuntu),
I solved it by editing the file:
/usr/bin/init-zram-swapping

in this way:

```
#!/bin/sh

modprobe zram

cat /sys/class/zram-control/hot_add
dev=$(ls /dev/zram* 2>/dev/null | wc -l)
dev=$(($dev-1))

# Calculate memory to use for zram (equal of ram)
totalmem=`LC_ALL=C free | grep -e "^Mem:" | sed -e 's/^Mem: *//' -e 's/  *.*//'`
mem=$((totalmem / 2 * 1024))

# initialize the devices
echo $mem > /sys/block/zram$dev/disksize
mkswap /dev/zram$dev
swapon -p 5 /dev/zram$dev
```

Now zramctl:
```
NAME       ALGORITHM DISKSIZE   DATA  COMPR  TOTAL STREAMS MOUNTPOINT
/dev/zram1 lzo-rle      13.2G     4K    74B    12K       4 [SWAP]
/dev/zram0 lz4             1G 384.7M 101.6M 105.6M       4 /var/log
```

By(t)e