azlux/log2ram #243: Unable to get log2ram working

Fadel-K, 2025-04-05:
I installed log2ram twice but its showing that it failed when I run `systemctl status log2ram`

 ```
log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; preset: enabled)
     Active: failed (Result: timeout) since Sat 2025-02-22 00:20:34 +03; 1 month 12 days ago
   Main PID: 383 (code=killed, signal=TERM)
        CPU: 22.734s

Feb 22 00:20:34 fadel-desktop log2ram[405]: mkdir: cannot create directory ‘/var/hdd.log’: Read-only file system
Feb 22 00:20:34 fadel-desktop log2ram[406]: mount: /var/hdd.log/: mount point does not exist.
Feb 22 00:20:34 fadel-desktop log2ram[406]:        dmesg(1) may have more information after failed mount system call.
Feb 22 00:20:34 fadel-desktop log2ram[408]: mount: /var/hdd.log/: mount failed: Unknown error 5005.
Notice: journal has been rotated since unit was started, output may be incomplete.
```

`journalctl -u log2ram -e `shows
```

Feb 22 00:20:34 fadel-desktop log2ram[405]: mkdir: cannot create directory ‘/var/hdd.log’: Read-only file system
Feb 22 00:20:34 fadel-desktop log2ram[406]: mount: /var/hdd.log/: mount point does not exist.
Feb 22 00:20:34 fadel-desktop log2ram[406]:        dmesg(1) may have more information after failed mount system call.
Feb 22 00:20:34 fadel-desktop log2ram[408]: mount: /var/hdd.log/: mount failed: Unknown error 5005.

```

azlux, 2025-04-05:
> Feb 22 00:20:34 fadel-desktop log2ram[405]: mkdir: cannot create directory ‘/var/hdd.log’: Read-only file system

Look like it's not a log2ram issue, your filesystem (at least `/var`) is in read-only mode.

Fadel-K, 2025-04-05:
Oh could you help me fix it? I was following tutorials on how to reduce SD card wear and tear. I might have done that accidently

azlux, 2025-04-07:
Your label are "weird", I don't use fstab like that.
What is the output of `mount`command ?

Tronicster, 2025-06-24:
im having a similar issue it says  log2ram.service - Log2Ram                                                                                               Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)                                Active: failed (Result: exit-code) since Tue 2025-06-24 15:00:52 EDT; 21s ago                                          Docs: https://github.com/azlux/log2ram                                                                            Process: 338 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)                                 Main PID: 338 (code=exited, status=1/FAILURE)                                                                             CPU: 301ms                                                                                                                                                                                                                        Jun 24 15:00:51 raspberrypi systemd[1]: Starting Log2Ram...                                                          Jun 24 15:00:52 raspberrypi log2ram[338]: ERROR: RAM disk for "/var/hdd.log/" too small. Can't sync.                 Jun 24 15:00:52 raspberrypi log2ram[338]: File(s) causing issues                                                     Jun 24 15:00:52 raspberrypi log2ram[338]: : 169M        /var/hdd.log/journal                                         Jun 24 15:00:52 raspberrypi log2ram[367]: /usr/local/bin/log2ram: line 72: mail -s Log2Ram Error on raspberrypi root>Jun 24 15:00:52 raspberrypi systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE          Jun 24 15:00:52 raspberrypi systemd[1]: log2ram.service: Failed with result 'exit-code'.                             Jun 24 15:00:52 raspberrypi systemd[1]: Failed to start Log2Ram.