azlux/log2ram #247: Is this how logs are supposed to look? -- Boot ca9164cc44384ec8aea2470b59e9681b --

webmogul1, 2025-06-26:
This is part of Proxmox System log.

```

Jun 26 11:23:43 pve01 kernel: vmbr2: port 8(veth180i0) entered disabled state
-- Boot ca9164cc44384ec8aea2470b59e9681b --
Jun 26 11:41:12 pve01 systemd[1]: Started session-13.scope - Session 13 of User root.
-- Boot 0fc3e6dca6aa4074a2a9def664299fb1 --
Jun 26 11:23:43 pve01 kernel: veth180i0 (unregistering): left allmulticast mode
-- Boot ca9164cc44384ec8aea2470b59e9681b --
Jun 26 11:41:12 pve01 login[21332]: ROOT LOGIN  on '/dev/pts/0'
-- Boot 0fc3e6dca6aa4074a2a9def664299fb1 --
Jun 26 11:23:43 pve01 kernel: veth180i0 (unregistering): left promiscuous mode
Jun 26 11:23:43 pve01 kernel: vmbr2: port 8(veth180i0) entered disabled state
Jun 26 11:23:43 pve01 kernel: vmbr2: port 3(veth110i0) entered disabled state
Jun 26 11:23:43 pve01 kernel: veth110i0 (unregistering): left allmulticast mode
Jun 26 11:23:43 pve01 kernel: veth110i0 (unregistering): left promiscuous mode
Jun 26 11:23:43 pve01 kernel: vmbr2: port 3(veth110i0) entered disabled state
Jun 26 11:23:43 pve01 systemd[1]: pve-container@160.service: Deactivated successfully.
```

[image]

azlux, 2025-08-13:
What is this ticket ? It's not related to log2ram.

doloopuntil, 2025-08-25:
I think @webmogul1  is onto something. After having installed log2ram on my Debian Trixie, I am getting a jumbled up system log
```
...
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: proc-sys-fs-binfmt_misc.mount: Deactivated successfully.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:     'statsd.bad_lines_seen': 0,
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Unset automount proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   counter_rates: {
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: proc-sys-fs-binfmt_misc.automount: Deactivated successfully.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   timer_data: {},
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Stopped systemd-binfmt.service - Set Up Additional Binary Formats.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   gauges: { 'statsd.timestamp_lag': 0 },
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: systemd-binfmt.service: Deactivated successfully.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   timers: {},
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Stopped systemd-timesyncd.service - Network Time Synchronization.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   },
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: systemd-timesyncd.service: Deactivated successfully.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:     'statsd.metrics_received': 0
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd-networkd[509]: enp7s0: DHCPv6 lease lost
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:     'statsd.packets_received': 0,
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd-networkd[509]: enp7s0: DHCP lease lost
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:     'statsd.bad_lines_seen': 0,
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Stopped systemd-tmpfiles-setup.service - Create System Files and Directories.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]:   counters: {
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: systemd-tmpfiles-setup.service: Deactivated successfully.
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]: {
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Stopping systemd-timesyncd.service - Network Time Synchronization...
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:14:53 fresh-footprints-main sh[1116]: Flushing stats at  Mon Aug 25 2025 05:14:53 GMT+0000 (Coordinated Universal Time)
-- Boot 0b9557ec5fed4c169b75fb363c1c4408 --
Aug 25 20:46:48 fresh-footprints-main systemd[1]: Stopping systemd-random-seed.service - Load/Save OS Random Seed...
-- Boot 22d2b5e791ea47efbad94344e0e24e24 --
Aug 25 05:05:01 fresh-footprints-main CRON[16104]: pam_unix(cron:session): session closed for user root
...
``` 
where log entries for different system boots (boots at different times) seem to be interleaved, instead of being grouped together.

JVKeller, 2025-09-15:
@azlux I had ran that command at one point to try and get things re synced. Whether that is what corrected it for me, I will never know. :) 

Maybe @webmogul1 can test that if he's still having issues.