azlux/log2ram #248: Request for advice on dedicated /var/log partition

antoinejdd, 2025-06-30:
Hi,

I've recently set up log2ram on my raspberry pi, thank you very much for your efforts!

Before the installation of log2ram, I've already partitioned my system with LVM. Here is my partitioning:
```
antoinejdd@brol:~ $ cat /etc/fstab 
PARTUUID=aa59b49a-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=aa59b49a-02  /               ext4    defaults,noatime  0       1

# custom setup
/dev/rpi_rootfs_vg/home /home ext4 noatime,nodev 0 2
/dev/rpi_rootfs_vg/var /var ext4 relatime 0 2
/dev/rpi_rootfs_vg/var_log /var/log ext4 noatime,nodev,nosuid,noexec 0 2
/dev/rpi_rootfs_vg/var_www /var/www ext4 noatime,nodev,nosuid 0 2
/dev/rpi_rootfs_vg/var_tmp /var/tmp ext4 noatime,nodev,nosuid 0 2
```
Somehow, it feels wrong to mount the `/dev/rpi_rootfs_vg/var_log` LV on /var/log. I was afraid of conflict with log2ram

But I didn't modify the `/etc/fstab` file following the install of log2ram.

After the log2ram install, here is how my storage volumes look like:
```
antoinejdd@brol:~ $ lsblk
NAME                      MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda                         8:0    0 465.8G  0 disk 
├─sda1                      8:1    0   512M  0 part /boot/firmware
├─sda2                      8:2    0  50.3G  0 part /
└─sda3                      8:3    0 321.9G  0 part 
  ├─rpi_rootfs_vg-home    254:0    0    10G  0 lvm  /home
  ├─rpi_rootfs_vg-var     254:1    0    10G  0 lvm  /var
  ├─rpi_rootfs_vg-var_log 254:2    0     5G  0 lvm  /var/hdd.log
  │                                                 /var/log
  ├─rpi_rootfs_vg-var_www 254:3    0     5G  0 lvm  /var/www
  └─rpi_rootfs_vg-var_tmp 254:4    0     1G  0 lvm  /var/tmp
zram0                     253:0    0   256M  0 disk /var/log
zram1                     253:1    0     3G  0 disk [SWAP]
antoinejdd@brol:~ $ df -h
Filesystem                         Size  Used Avail Use% Mounted on
udev                               1.6G     0  1.6G   0% /dev
tmpfs                              380M  1.4M  379M   1% /run
/dev/sda2                           50G  2.7G   45G   6% /
tmpfs                              1.9G     0  1.9G   0% /dev/shm
tmpfs                              5.0M   16K  5.0M   1% /run/lock
tmpfs                              950M     0  950M   0% /tmp
/dev/mapper/rpi_rootfs_vg-home     9.8G  662M  8.6G   7% /home
/dev/mapper/rpi_rootfs_vg-var      9.8G  1.8G  7.6G  19% /var
/dev/sda1                          510M   63M  448M  13% /boot/firmware
/dev/mapper/rpi_rootfs_vg-var_tmp  974M  328K  906M   1% /var/tmp
/dev/zram0                         224M   87M  119M  43% /var/log
/dev/mapper/rpi_rootfs_vg-var_www  4.9G  1.9G  2.8G  41% /var/www
/dev/mapper/rpi_rootfs_vg-var_log  4.9G  105M  4.5G   3% /var/hdd.log
tmpfs                              380M     0  380M   0% /run/user/17049
```

Should I adapt my `/etc/fstab` to mount the `/dev/rpi_rootfs_vg/var_log` LV on `/var/hdd.log` mountpoint instead? Or does my current setup look OK and I should let it as it currently is?

Thanks in advance

azlux, 2025-06-30:
Both will be ok, log2ram is a simple bind mount.
With multiple mount for the same location isn't the best for debug, but should be fine.

antoinejdd, 2025-07-17:
Sorry for the late answer.

Alright, thanks! If you have suggestion to better set this up, don't hesitate to tell me. In the meantime, I'll close this issue.